open62541/open62541 #6340: Call to UA_Client_connect never returns when server is not running

Ry-Kode, 2024-03-10:
Hello,

When I call "`UA_Client_connect`" or "`UA_Client_connectUsername`" to connect to a non-existing server, the call never returns.
I tried with different config timeout settings from 500ms up to 15000ms.

Version: v1.3.8
OS: Ubuntu 22.04.4 LTS
CMake options:
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUND
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-11
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-11
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
CMAKE_INSTALL_LOCALEDIR:PATH=
CMAKE_INSTALL_LOCALSTATEDIR:PATH=var
CMAKE_INSTALL_MANDIR:PATH=
CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/include
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_INSTALL_RUNSTATEDIR:PATH=
CMAKE_INSTALL_SBINDIR:PATH=sbin
CMAKE_INSTALL_SHAREDSTATEDIR:PATH=com
CMAKE_INSTALL_SYSCONFDIR:PATH=etc
CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/gmake
CMAKE_MODULE_LINKER_FLAGS:STRING=
CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_NM:FILEPATH=/usr/bin/nm
CMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopy
CMAKE_OBJDUMP:FILEPATH=/usr/bin/objdump
CMAKE_RANLIB:FILEPATH=/usr/bin/ranlib
CMAKE_READELF:FILEPATH=/usr/bin/readelf
CMAKE_SHARED_LINKER_FLAGS:STRING=
CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
CMAKE_STATIC_LINKER_FLAGS:STRING=
CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_STRIP:FILEPATH=/usr/bin/strip
CMAKE_VERBOSE_MAKEFILE:BOOL=FALSE
DVIPDF_CONVERTER:FILEPATH=/usr/bin/dvipdf
DVIPS_CONVERTER:FILEPATH=DVIPS_CONVERTER-NOTFOUND
GIT_EXECUTABLE:FILEPATH=/usr/bin/git
HTLATEX_COMPILER:FILEPATH=HTLATEX_COMPILER-NOTFOUND
LATEX2HTML_CONVERTER:FILEPATH=LATEX2HTML_CONVERTER-NOTFOUND
LATEX_COMPILER:FILEPATH=LATEX_COMPILER-NOTFOUND
LUALATEX_COMPILER:FILEPATH=LUALATEX_COMPILER-NOTFOUND
MAKEINDEX_COMPILER:FILEPATH=MAKEINDEX_COMPILER-NOTFOUND
PDFLATEX_COMPILER:FILEPATH=PDFLATEX_COMPILER-NOTFOUND
PDFTOPS_CONVERTER:FILEPATH=/usr/bin/pdftops
PS2PDF_CONVERTER:FILEPATH=/usr/bin/ps2pdf14
Python3_EXECUTABLE:FILEPATH=/usr/bin/python3
SPHINX_EXECUTABLE:FILEPATH=SPHINX_EXECUTABLE-NOTFOUND
UA_AMALGAMATION_MULTIARCH:BOOL=OFF
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_FUZZING:BOOL=OFF
UA_BUILD_FUZZING_CORPUS:BOOL=OFF
UA_BUILD_OSS_FUZZ:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_DEBUG:BOOL=OFF
UA_DEBUG_DUMP_PKGS:BOOL=OFF
UA_DEBUG_FILE_LINE_INFO:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_CERT_REJECTED_DIR:BOOL=OFF
UA_ENABLE_CLANG_COV:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DEBUG_SANITIZER:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_IMMUTABLE_NODES:BOOL=OFF
UA_ENABLE_INLINABLE_EXPORT:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MQTT:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESETLOADER:BOOL=OFF
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_BUFMALLOC:BOOL=OFF
UA_ENABLE_PUBSUB_ENCRYPTION:BOOL=OFF
UA_ENABLE_PUBSUB_FILE_CONFIG:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_MONITORING:BOOL=OFF
UA_ENABLE_PUBSUB_SKS:BOOL=OFF
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_XML_ENCODING:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_WERROR:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=100
UA_NAMESPACE_ZERO:STRING=REDUCED
UA_NODESET_DIR:STRING=/home/vanhoj14/libs/open62541/deps/ua-nodeset
UA_PACK_DEBIAN:BOOL=OFF
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND

**code snippet:**

```
 //-- CONNECT
  printf("Try to create secure connection \n");
  if (lEnablePasswordLogin) {
    printf("Password login is ENABLED \n");
    retVal = UA_Client_connectUsername(GCOPCUAClient, endpointUrl, "admin", "admin");

  } else {
    printf("Password login is DISABLED \n");
    retVal = UA_Client_connect(GCOPCUAClient, endpointUrl);
  }

  printf("Login procedure Finished \n");          // -> NEVER GETS HERE WHERE SERVER IS NOT RUNNING

```

**console output:**
```
...
Setting Default Encryption
[2024-03-10 12:47:18.509 (UTC+0100)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2024-03-10 12:47:18.512 (UTC+0100)] info/securitypolicy        The basic256 security policy with openssl is added.
[2024-03-10 12:47:18.514 (UTC+0100)] info/securitypolicy        The Aes256Sha256RsaPss security policy with openssl is added.
[2024-03-10 12:47:18.516 (UTC+0100)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2024-03-10 12:47:18.517 (UTC+0100)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
Default Encryption finished
Try to create secure connection
Password login is DISABLED
[2024-03-10 12:47:18.519 (UTC+0100)] info/eventloop     Starting the EventLoop
[2024-03-10 12:47:18.519 (UTC+0100)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-03-10 12:47:18.521 (UTC+0100)] debug/network      TCP     | Open a connection to "PMSVM" on port 53530
[2024-03-10 12:47:18.522 (UTC+0100)] info/network       TCP 4   | New connection to "PMSVM" on port 53530
[2024-03-10 12:47:18.522 (UTC+0100)] debug/client       Client Status: ChannelState: Connecting, SessionState: Closed, ConnectStatus: Good
[2024-03-10 12:47:18.522 (UTC+0100)] debug/network      TCP 4   | Activity on the socket
[2024-03-10 12:47:18.522 (UTC+0100)] info/network       TCP 4   | The connection closes with error 111
[2024-03-10 12:47:18.522 (UTC+0100)] debug/network      TCP 4   | Shutdown triggered
[2024-03-10 12:47:18.522 (UTC+0100)] debug/eventloop    TCP 4   | Delayed closing of the connection
[2024-03-10 12:47:18.522 (UTC+0100)] info/channel       TCP 0   | SC 0  | Could not open the connection
[2024-03-10 12:47:18.522 (UTC+0100)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionRejected
[2024-03-10 12:47:18.522 (UTC+0100)] info/network       TCP 4   | Socket closed
// HANGS HERE INDEFINITLY

```

Similar issue I have found:
[https://github.com/open62541/open62541/issues/6071](url)

thanks!

dodissimo77, 2024-05-19:
There is an error in  connectSync(UA_Client *client) :
 
    /* EventLoop is started. Otherwise initConnect would have failed. */
    UA_EventLoop *el = client->config.eventLoop;
    UA_assert(el);
    UA_DateTime now = el->dateTime_nowMonotonic(el);
    UA_DateTime maxDate = now + ((UA_DateTime)client->config.timeout * UA_DATETIME_MSEC);

    /* Initialize the connection */
    initConnect(client);
    notifyClientState(client);
    if(client->connectStatus != UA_STATUSCODE_GOOD)
        return;

the two blocks of codes are swapped.. the event loop is not already started.
This is the correct sequence : 

    /* Initialize the connection */
    initConnect(client);
    notifyClientState(client);
    if(client->connectStatus != UA_STATUSCODE_GOOD)
        return;

    /* EventLoop is started. Otherwise initConnect would have failed. */
    UA_EventLoop *el = client->config.eventLoop;
    UA_assert(el);
    UA_DateTime now = el->dateTime_nowMonotonic(el);
    UA_DateTime maxDate = now + ((UA_DateTime)client->config.timeout * UA_DATETIME_MSEC);

paperino1983, 2024-10-04:
Hi [dodissimo77], I confirm it works your fix. Thanks !

mariusebastian, 2024-11-01:
Hello. I had the same problem and the fix works for the case where server is not running. However, if the client is configured to connect to an endpoint url in the wrong ip range, the problem persists, ie UA_Client_connect never returns.
Terminal output:
``` 
[2024-11-01 08:22:21.430 (UTC+0000)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-11-01 08:22:21.430 (UTC+0000)] info/network       TCP 6   | Opening a connection to "10.11.0.196" on port 4840
[2024-11-01 08:22:26.435 (UTC+0000)] error/client       The connection has timed out before it could be fully opened
[2024-11-01 08:22:26.435 (UTC+0000)] info/channel       TCP 6   | SC 0  | SecureChannel closed
[2024-11-01 08:22:26.435 (UTC+0000)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadTimeout
[2024-11-01 08:22:26.435 (UTC+0000)] info/network       TCP 6   | Socket closed

o6-nbeier, 2026-10-20:
Thank you for the report.

The connect order and the EventLoop clock initialization were fixed in 352c729 and 452696c, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier