open62541/open62541 #6343: Porting to new OS/platforms is not documented in version 1.4

RavilN, 2024-03-12:
## Description

The existing version of the documentation (https://www.open62541.org/doc/master/building.html#building-for-specific-architectures) refers to platform-specific files located under the sub-folder **arch**, but this folder does not exist anymore in version 1.4. 
Moreover, files that were used to port to certain platforms, for example, the **freertosLWIP** folder, do not exist anymore.

Probably now all platform-specific code should be under folder **plugins**, but there is no information about this in the available documentation.

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.4.0-rc2

RavilN, 2024-03-14:
I can see that version v1.4.0-rc2 has a folder ``arch`` and ``arch/Readme.md`` file describing how to port. It is the current dev version of the ``master `` branch does not have this folder and readme file. 
So now the question is: is the file ``arch/Readme.md`` in version v1.4.0-rc2 still up to date, it was deleted accidentally later, or it is now outdated and needs to be replaced by a new documentation file?

jpfr, 2024-04-03:
This is now back on master and updated.

closed 2024-04-03 by jpfr