open62541/open62541 #6366: Creating tcp socket beside a opc ua server

muurc, 2024-03-27:
## Description
Currently, I am working on a project in which my server gets TCP packages from another TCP server and displays the data on an information model of the OPCUA server. So far, I've tried to implement an extra TCP socket with the arpa/inet header and maintain the management in another pthread.  However, it seems to shut down the main event loop of the OPCUA server and stop the server. It can create the tcp socket, yet there is no connection to the server. It then seems to jump back to the OPCUA event loop and stop the socket there.

Is there even a way to implement this TCP client plugin alongside the OPCUA server? If yes, should I use the arpa/inet structure, or is there maybe a way to use the OPCUA library for this? And perhaps if this is not makeable, is there any other way you could recommend me doing to receive TCP packages?

I appreciate every comment.

## Background Information / Reproduction Steps
Creating a OPCUA server, then implementing a arpa socket and running the thread in a pthread.

Used CMake options:

```bash
cmake 
-DBUILD_SHARED_LIBS=ON 
-DCMAKE_BUILD_TYPE=RelWithDebInfo 
-DUA_NAMESPACE_ZERO=FULL 
-UA_ENABLE_MULTITHREAD=ON ..
```

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): 1.4.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Ubuntu 20.04.6
 - [ ]
[image]
 Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-04-02:
Yes. The EventLoop was made exactly for that.
https://github.com/open62541/open62541/blob/master/include/open62541/plugin/eventloop.h

Every server and client configuration contains an EventLoop.
You can interact with that and run your socket in parallel to the normal server operations.

You can reopen this issue if there are further questions.

closed 2024-04-02 by jpfr

muurc, 2024-04-04:
Hello Mister Pfrommer,

I hope you dont mind me asking, but I was wondering if it is possible to
send "normal" tcp messages from an open62541 client to a server, and what feature would
I need for implementing a message handler on the server side.

I would appreciate a suggestion from your side.

Sincerely,

Marc Futschik

https://github.com/open62541/open62541/blob/master/include/open62541/plugin/eventloop.h

jpfr, 2024-04-04:
Only TCP would not be OPC UA.
But if you need it, check the examples here: https://github.com/open62541/open62541/blob/master/tests/check_eventloop_tcp.c
You can do the same with the EventLoop that is driving every open62541 client and server.
Basically you open a connection with a callback method that gets triggered when an event happens (message arrives, connection closes, etc.).

muurc, 2024-04-05:
Thank you very much again. I have been trying to find the right solution for a whole day now.
So your showing me the correct approach certainly helped me.

Have a nice day, and weekend!

Marc Futschik