open62541/open62541 #6388: Sporadic connection failures without TCP reuseaddr option

keba-estr, 2024-04-08:
## Description
With commit 3bc62f7812b835285bb184d7aa0c09d0c4293efe (PR #[6255](https://github.com/open62541/open62541/pull/6255))  the default value of the socket option REUSE_ADDR was changed to disabled.

If this default value is used, sporadic connection error occurs in tests despite the fact that both the server and client are restarted between each test case.

I assume that sockets are closed when the server process is stopped. Is there any timeout in linux that prevents the socket to be reused after restarting the server process?

## Background Information / Reproduction Steps
Return code of UA_Client_connect(): Bad connection rejected:0x80ac0000

Traces on server side:
TCP	| Listening on all interfaces
TCP 82	| Creating server socket for "0.0.0.0" on port 4842
TCP 82	| Error binding the socket to the address 0.0.0.0 (Address already in use)
TCP 82	| Creating server socket for "::" on port 4842

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.4
 - [ ] Operating system: Linux

jpfr, 2024-04-08:
We now have a configuration option in client/server to enable the reuseaddr option.
There problems in the certification cycle because our system did not complain when several servers were started on the same port...

keba-estr, 2024-04-08:
Is it secure enough to enable the reuseaddr option in the server by default, or is this a security issue?

jpfr, 2024-04-08:
Shouldn't be a problem.

closed 2024-04-08 by keba-estr