open62541/open62541 #6390: Custom DataTypes don't show members in client

rafaeljob, 2024-04-08:
## Description
I'm not able to properly display the members (structure fields) of a custom DataType which may be the reason on why writings are also not working.
## Background Information / Reproduction Steps
I have followed the example provided under `master\examples\custom_datatype` in which I have successfully created the `DataType` node, `VariableType` node and the `Variable` node. These are displayed under UA Expert.

As presented in the picture below, the `DataTypeDefinition` does not show the `StructureDefinition`.

[image]

[image]

Is there something missing in the example provided ? Or is it a limitation of UA Expert ?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4.0-rc2
 - [x] Other OPC UA SDKs used (client or server): UA Expert
 - [x] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2024-09-10:
May define UA_ENABLE_TYPEDESCRIPTION is not enabled.

ayarikba, 2024-10-17:
Hello i have similar problem. Even if UA_ENABLE_TYPEDESCRIPTION is enabled, i can't see member of structure using UA_Expert. Is there any solution?