open62541/open62541 #6392: Server crash when browsing ServerDiagnostics object

tboeckel, 2024-04-08:
## Description

I built open62541 with UA_ENABLE_DIAGNOSTICS. The Server object contains the ServerDiagnostics object. I can open the variable tree with UaExpert and inspect every single contained variable, except the SessionsDiagnosticsSummary objects of a local connection to that server. In that case the server application crashes with "UA_LOCK_ASSERT: Assertion `lock->mutexCounter == num' failed" as final output. The SessionsDiagnosticsSummary objects of the UaExpert connection can be inspected without any problem. It is only the SessionsDiagnosticsSummary of our own client (also open62541 based), which cannot be inspected.

I already tried to find the exact location of the assert call, but unfortunately gdb's stack trace seems to be corrupt and reveals no valuable information. So it could be any of the 72 occurances of UA_LOCK_ASSERT().

Is there anything special which a client must do to support reading its own session information by other clients (i.e. UaExpert) in a foreign server?

Used CMake options:

CMAKE_CXX_COMPILER=g++
CMAKE_C_COMPILER=gcc
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OPENSSL
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_MULTITHREADING:STRING=100
OPEN62541_VERSION=v1.3.9

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.3.9
 - [X] Other OPC UA SDKs used (client or server): UaExpert 1.7.1
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tboeckel, 2024-04-08:
To add some more details: it is sufficent to click on one of the 3 objects "SessionDiagnostics", "SessionSecurityDiagnostics" or "SubscriptionDiagnosticsArray" in the object tree to make the server crash. Doing the same for the UaExpert connection's variables just produces lots of "NodeClasss of ns=1;i=50703 has bad statuscode: BadNodeIdUnknown" log entries, but it doesn't make the server crash.

jpfr, 2024-04-08:
This is probably already fixed in the 1.3 branch.
I just cut a new release. Please test.
https://github.com/open62541/open62541/releases/tag/v1.3.10

tboeckel, 2024-04-09:
Almost...

I just upgraded to 1.3.10.

Clicking the 3 objects "SessionDiagnostics", "SessionSecurityDiagnostics" or "SubscriptionDiagnosticsArray", as well as all subobjects in "SessionDiagnostics" and "SessionSecurityDiagnostics". But clicking on the "1" subobject in "SubscriptionDiagnosticsArray" still crashes the server. However, I can drag the "SubscriptionDiagnosticsArray" object into the Data Access View and double click on the "Value" column to show all embedded variables without any crash. I can even just open the "1" object node in the object tree and click on any object inside the "1" object, but a direct click on the "1" object still causes a crash.

The last message in the console is still "UA_LOCK_ASSERT: Assertion `lock->mutexCounter == num' failed."

I will also try release 1.4.0 now.

jpfr, 2024-05-13:
Fixed via #6468.

closed 2024-05-13 by jpfr

andyfu6, 2025-04-01:
> Fixed via [#6468](https://github.com/open62541/open62541/pull/6468).

So, which version can be used? We just used open62541-1.3.6, seems have the same error.