open62541/open62541 #6395: Version 1.4 not buildable with debian buster

keba-estr, 2024-04-09:
## Description

libopen62541 needs at least libmbedtls-dev (>= 2.28.0) in the control-template. Debian buster only has version 2.16.0-1.

Dependency to libmbedtls is only done for libmbedtls14, not libmbedtls12 and libmbedtls10 any more

Due to Commit: 9e0b766fbba73be8d703b6ccbfdeec5bfd200b48 (PR #6369)

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

keba-estr, 2024-04-09:
@jpfr: Why is debian buster not supported any more?

jpfr, 2024-04-09:
You are right. Buster is two versions behind Debian stable.
But it still receives Long-Term-Support.
The dependency can be reduced to older versions of mbedTLS.

keba-estr, 2024-04-09:
Compilable for debian buster with libmbedtls-dev (>= 2.16.0) and dependency to libmbedtls12 | libmbedtls14

jpfr, 2024-04-17:
Fixed on 1.4.

closed 2024-04-17 by jpfr