open62541/open62541 #6396: Cannot set Application Server Uri to use self-signed Certificate

giuseppedevito, 2024-04-09:
## Description
I have generate custom certificates to use in my OPCUA server implementation.
In my server certificate I have the custom alternative name

```
X509v3 Subject Alternative Name: 
                DNS:opcua.dragoon.gdev.it
```
Obviously I translated server cert and key in DER format; the same for the CA cert.
In the server code (the main() function) I manipulate the configuration as follows:

```
UA_String_clear(&uaServerConfig->applicationDescription.applicationUri);
UA_LocalizedText_clear(&uaServerConfig->applicationDescription.applicationName);
uaServerConfig->applicationDescription.applicationUri = UA_String_fromChars("DNS:opcua:dragoon.gdev.it");
uaServerConfig->applicationDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en-US", "VMS OPC UA Server Application");
```

but when I call

```
UA_ServerConfig_setDefaultWithSecureSecurityPolicies(uaServerConfig, 4840,
       &certificate, &privateKey, trustList, trustListSize, issuerList, issuerListSize,
               revocationList, revocationListSize);
```

the custom app description info ("DNS:opcua:dragoon.gdev.it") will be overwritten by the Open62541 library that is
calling setDefaultConfig and, in particular, the code (present in ua_config_default.c):

```
UA_ApplicationDescription_clear(&conf->applicationDescription);
conf->applicationDescription.applicationUri = UA_STRING_ALLOC(APPLICATION_URI_SERVER);
conf->applicationDescription.productUri = UA_STRING_ALLOC(PRODUCT_URI);
```

Is there a way to change applicationDescription.applicationUri at init time without it be changed later?
I can change the macros 

```
#define MANUFACTURER_NAME "open62541"
#define PRODUCT_NAME "open62541 OPC UA Server"
#define PRODUCT_URI "http://open62541.org"
#define APPLICATION_NAME "open62541-based OPC UA Application"
#define APPLICATION_URI "urn:unconfigured:application"
#define APPLICATION_URI_SERVER "urn:open62541.server.application"
```

in  ua_config_default.c, but I think it will be overwritten by library updates.

## Background Information / Reproduction Steps

See description

Used CMake options:

```bash
-DCMAKE_BUILD_TYPE=MinSizeRel -DUA_LOGLEVEL=99 -DBUILD_SHARED_LIBS=yes 
-DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON 
-DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS=ON -DUA_ENABLE_ENCRYPTION=OPENSSL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4
 - [x] Other OPC UA SDKs used (client or server): UAExpert
 - [x] Operating system: Debian 12
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-04-11:
Sure. Just update the application description to reflect the certificate before you start the server.

o6-nbeier, 2026-10-20:
Thank you for the question.

Call the default-security helper first, then set the application URI and name before starting the server, and use a URI SAN (not a DNS entry) in the certificate.

Closing for now.

closed 2026-10-20 by o6-nbeier