open62541/open62541 #6400: Version 1.4 not linkable with debian linux

keba-estr, 2024-04-10:
## Description
Linking the open62541 does not work any more on linux systems, as the files libopen62541.a and libopen62541.so have been removed from the dev debian package.

Changed in commit https://github.com/open62541/open62541/commit/9e0b766fbba73be8d703b6ccbfdeec5bfd200b48 (PR https://github.com/open62541/open62541/pull/6369)

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-04-17:
I checked locally and it works.
1. Create the debian build environment with /tools/prepare_packaging.py
2. Build with dpkg-buildpackage -B

The generated package does contain the .so file.

jpfr, 2024-04-17:
The .so is now in the libopen62541 package.
The libopen62541-dev package depends on that.

So other packages can use our library without having to install the headers.

closed 2024-04-17 by jpfr