open62541/open62541 #6401: Error building nodesetloader in Open62541 library version 1.4.0.

xydan83, 2024-04-10:
## Description
Hi!
I'm having trouble building the new Open62541 1.4.0 version along with nodesetloader. Please see the screenshot below.

[image]

I noticed the same errors in the git autobuild system.
[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.0
 - [x] Operating system: Linux Ubuntu 22.04

matkonnerth, 2024-04-13:
here is the fix: https://github.com/open62541/open62541-nodeset-loader/pull/257

keba-estr, 2024-04-18:
Is the update of the open62541-nodeset-loader done automatically on the open62541 1.4 (or rather 1.4.1 hotfix) branch or do we need a PR?

matkonnerth, 2024-04-18:
PR is needed

keba-estr, 2024-04-19:
@jpfr: Please start an update with dependabot[bot] (however this works)

GoetzGoerisch, 2024-04-19:
Here you go: #6422

matkonnerth, 2024-05-03:
Is this issue resolved?

paddor, 2024-05-03:
I think it is

keba-estr, 2024-06-03:
@jpfr: please close

closed 2024-06-03 by jpfr