open62541/open62541 #6403: Custom DataType array items changing to ExtensionObject

andremoraisp, 2024-04-10:
## Description

When I have a custom type `Gift` and publish 2 variable nodes, one being a scalar and the other an array, I get two different behaviors after editing the value in UAExpert:

- scalar node: the value received in the `UA_ValueCallback.onWrite` is of type `Gift`
- array node: the value received in the `UA_ValueCallback.onWrite` is of type `ExtensionObject`

Is this difference intended? Should I check the type of the value every time I read the node, even though I've set the type of the node to be `Gift`?

I've read that _non-built-in_ types should always be wrapped in an `ExtensionObject`, but I've also read that open62541 should handle that transparently, as seen [in the docs](https://www.open62541.org/doc/1.3/types.html#variant):

> Variant
Variants may contain values of any type together with a description of the content. See the section on Generic Type Handling on how types are described. The standard mandates that variants contain built-in data types only. If the value is not of a builtin type, it is wrapped into an ExtensionObject. **open62541 hides this wrapping** transparently in the encoding layer. If the data type is unknown to the receiver, the variant contains the original ExtensionObject in binary or XML encoding.

## Background Information / Reproduction Steps

Example code here: https://gist.github.com/andremoraisp/16e14b8e696d055bcbf4ffc8713cd190

I ran the above example code, which creates a server with 2 nodes of the type `Gift`, 1 scalar and 1 array.
This code resets the values of the nodes once every 5 seconds.
It also register callbacks to write actions, printing the node type name.

The node value types are always `Gift`, until I change the values in UAExpert.
Then, while the scalar node keeps having a `Gift` value, the array node's value changes to an `ExtensionObject` type.

This can be seen in the logs.
Full logs (including open62541 generated) : [20240410.log](https://github.com/open62541/open62541/files/14937068/20240410.log)

```text
### Node Gift ARRAY, new value of type Gift, productId = 0
### Node Gift SCALAR, new value of type Gift, productId = 43
...
### Node Gift ARRAY, new value of type Gift, productId = 0
### Node Gift SCALAR, new value of type Gift, productId = 44
...
### Node Gift SCALAR, new value of type Gift, productId = 10
>>> Here I've set the scalar value on UAExpert
...
### Node Gift ARRAY, new value of type ExtensionObject, productId = -1
>>> Here I've set the array value[0] on UAExpert
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.9
 - [x] Other OPC UA SDKs used (client or server): UAExpert
 - [x] Operating system: Windows 11 Enterprise 22H2
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-04-10:
This was changed for the 1.4 release family.
Now we also unpack arrays of ExtensionObjects if and only if all members of the array are of the same type.

Since this is a breaking change, this will stay as-is for older release families.

andremoraisp, 2024-04-10:
Thank you for your response. I'm glad to know that this improvement is already available.

closed 2024-04-10 by andremoraisp