open62541/open62541 #6413: QUESTION - How to add args to callback function

EusebioAlsoliman, 2024-04-15:
I'm trying to use a callback similar to the one in https://www.open62541.org/doc/v1.4.0/tutorial_server_datasource.html#variable-value-callback

In the example, the function afterWriteTime() is static and also doesn't need additional arguments or data. However, in my case, I need to pass extra arguments and make calls to functions. How can I do that?

Thanks in advance.

ccvca, 2024-04-16:
You may either use the nodeContext or the sessionContext for this. Own arguments are not possible in C.

EusebioAlsoliman, 2024-04-16:
> You may either use the nodeContext or the sessionContext for this. Own arguments are not possible in C.

From the example I have seen that the callback functions should have this structure:

```
afterWriteTime(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeId, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data){
               // Code to be executed
}
```

But I don't know exactly how I can get the data from that nodeContext pointer. Also, how UA_Server_setNodeContext() really works?

Thanks in advance.

ccvca, 2024-04-17:
Use `UA_Server_setNodeContext` to set a pointer, which then can be used in the callback again:

```
// your Initialization

myVariable = (mytype*) malloc(sizeof(mytype));
UA_Server_setNodeContext(pServer, nodeId, myVariable);

afterWriteTime(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeId, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data){
               variableFromContext = (mytype*) nodeContext;

               // Code to be executed
}
```
PS: All NULL-Checks are missing!

EusebioAlsoliman, 2024-04-22:
> Use `UA_Server_setNodeContext` to set a pointer, which then can be used in the callback again:
> 
> ```
> // your Initialization
> 
> pMyVariable = (mytype*) malloc(sizeof(mytype));
> UA_Server_setNodeContext(pServer, nodeId, pMyVariable);
> 
> 
> afterWriteTime(UA_Server *server,
>                const UA_NodeId *sessionId, void *sessionContext,
>                const UA_NodeId *nodeId, void *nodeContext,
>                const UA_NumericRange *range, const UA_DataValue *data){
>                /// Get your variable again
>                pVariableFromContext = (mytype*) nodeContext;
> 
>                // Code to be executed
> }
> ```
> 
> PS: All NULL-Checks are missing!

Ok, now I've understood how it works, thanks.

closed 2024-04-22 by EusebioAlsoliman