open62541/open62541 #6414: 1.4 Branch: Failure to build on FreeBSD

thomvet, 2024-04-16:
I am trying to build binaries of open62541 on various platforms, so that they are usable from within the programming language Julia. The effort is documented here: https://github.com/JuliaPackaging/Yggdrasil/pull/7889

This worked well for v1.3.9 on all platforms considered (https://github.com/JuliaPackaging/Yggdrasil/pull/7783). However, when applying the same approach with v1.4, the build fails with FreeBSD with the following error message (full error log: https://buildkite.com/julialang/yggdrasil/builds/9637#018ee15a-1b7a-40cb-a499-14d13b648797):
```
[10:44:59] /workspace/srcdir/open62541/arch/eventloop_posix.c:551:46: error: use of undeclared identifier 'SO_REUSEPORT'
[10:44:59]     res \|= UA_setsockopt(sockfd, SOL_SOCKET, SO_REUSEPORT,
```

I am ccing @herrhotzenplotz because who is maintaining the FreeBSD port of open62541 (https://www.freshports.org/devel/open62541/)  

Specifically, we are wondering whether our compile problem here stems from this: 
https://github.com/JuliaPackaging/Yggdrasil/pull/7889#issuecomment-2056497595

I must admit that I don't know the open62541 code base (nor FreeBSD for that matter) well enough to help further here. But maybe anyone more knowledgeable could help out?

herrhotzenplotz, 2024-04-16:
Usually these compilation failures on FreeBSD are related to `_XOPEN_SOURCE` being forced to some value. POSIX does not define any socket functions and therefore FreeBSD hides them when `_XOPEN_SOURCE` is being defined. I will investigate.

herrhotzenplotz, 2024-04-16:
Indeed: Commenting out the `#define _XOPEN_SOURCE 600` in `include/open62541/config.h.in` and adding an include for `netinet/in.h` in `arch/posix/ua_architecture.h` will make it compile.

thomvet, 2024-04-16:
Wow. That was fast. Do you think you can PR this?

herrhotzenplotz, 2024-04-16:
> Wow. That was fast. Do you think you can PR this?

Sure! I will have to make changes anyways in order to update the port.

closed 2024-04-28 by jpfr