open62541/open62541 #6415: PubSub UDP interface name

keba-estr, 2024-04-16:
## Description
PubSub configuration fails with UDP interface name "eth1".

Configuration:
PubSub connection with connection address: URL: opc.udp://239.0.0.1:4840 NetworkInterface: eth1

Trace:
UDP   | Interface configuration preparation failed.
(code: inet_pton failed in eventloop_posix_udp.c line 119)

What should be the correct configuration in the network interface string, an IP address or the name of the network in the operating system?

## Background Information / Reproduction Steps
Works if NetworkInterface ist set to IP-Address of interface, e.g. 192.168.10.1

Used CMake options:

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

keba-estr, 2024-04-16:
UA_inet_pton was integrated in UDP event loop handling with commit 961ac08b0272da8ee9fa0c176c5e99e645d2e2c1

keba-estr, 2024-04-16:
Unicast configuration with  URL: opc.udp://192.168.10.1:4840 and NetworkInterface: eth1 is possible

keba-estr, 2024-04-16:
Proposal:

eventloop_posix_udp.c:

```
/**
 * Get ip address from network identifier (such as "eth0" or "eth0:0")
 *
 * @return -1 upon error, 0 upon success
 */
static int getIPv4(const char *pEthIf, struct in_addr *pIpv4) {

    struct ifreq ifc;
    int res;
    int sockfd;

    if((pEthIf == NULL) || (pIpv4 == NULL))
        return -1;

    sockfd = socket(AF_INET, SOCK_DGRAM, 0);
    if(sockfd < 0)
        return -1;

    strcpy(ifc.ifr_name, pEthIf);
    res = ioctl(sockfd, SIOCGIFADDR, &ifc);
    close(sockfd);
    if(res < 0)
        return -1;     

    *pIpv4 = ((struct sockaddr_in*)&ifc.ifr_addr)->sin_addr;
    return 0;
}
```

function setupMulticastRequest:

```
        if(UA_inet_pton(AF_INET, interfaceAsChar, &req->ipv4.imr_interface) <= 0) {
            /* try to convert interface name to ip address */
            if(getIPv4(interfaceAsChar, &req->ipv4.imr_interface) < 0) {
                UA_LOG_ERROR(logger, UA_LOGCATEGORY_SERVER,
                             "UDP\t| Interface configuration preparation failed.");
                return UA_STATUSCODE_BADINTERNALERROR;
            }
        }
```

jpfr, 2024-04-16:
I think it can be done with the interface index similar to the IPv6 code path.

On Linux/BSD there is `ip_mreqn` (https://man.openbsd.org/ip.4) and on Windows you can use `0.0.0.1` for the `imr_interface` to indicate the network interface with index 1 (https://learn.microsoft.com/en-us/windows/win32/api/ws2ipdef/ns-ws2ipdef-ip_mreq).

jpfr, 2024-04-16:
Good find!

keba-estr, 2024-04-17:
See PR #6419

closed 2024-04-17 by keba-estr