open62541/open62541 #6420: Problems when enabling encryption as flag in cmake command

muurc, 2024-04-17:
## Description
I have troubles enabling encryption with openssl when building the open62541 lib. It doesnt show me any error when running the cmake command itself, however, when I am calling the make command the terminal says pretty fast: 

_ua_eventfilter_grammar.leg:30343:1: error: conflicting types for ‘UA_OpenSSL_LoadLocalCertificate’_ 

OpenSSL is installed(libssl-dev), I am using ubuntu 20.04.6 LTS.   

## Background Information / Reproduction Steps

Configuring building flags

Used CMake options:

```bash
cmake 
-DCMAKE_BUILD_TYPE=RelWithDebInfo 
-DUA_ENABLE_NODEMANAGEMENT=ON 
-DUA_ENABLE_ENCRYPTION=ON 
-DUA_ENABLE_ENCRYPTION_OPENSSL=ON 
-DUA_MULTITHREADING=ON 
-DUA_NAMESPACE_ZERO=FULL 
-DUA_ENABLE_AMALGAMATION=ON 
-DUA_ENABLE_SUBSCRIPTIONS=ON 
-DUA_ENABLE_PUBSUB=ON 
-DUA_ENABLE_MULTITHREADING=ON ..

```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04.06 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-04-17:
Something is very wrong here.
The file `ua_eventfilter_grammar.leg` does not have 30k lines.

jpfr, 2024-04-23:
I think you need to add -lx509…

muurc, 2024-04-23:
I rebuilt the project and the issue was solved after I added the correct parameters for OpenSSL. I woul have another question although, I followed the example of "server_encryption.c". However the terminal still tells the following: 
**x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.**

So my question is, didnt I corretly added the policy, and if not how can I add it?