open62541/open62541 #6437: No return value from server-method and status code *uncertain*

augustinust, 2024-04-23:
## Description
I have a simple server with some methods. If a method fails(not completely), I want to provide 
some more information beside the status code to the client.

When I use *UA_STATUSCODE_UNCERTAIN* as method return code, nothing (except the status code)
is returned to the client.

Is this behaviour correct?

If so, what is the "correct" way to indicate something went wrong executing the method?

I thought of returning a UA_Variant with
- status code
- the explanation in case of a failure

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag):  1.3
 - [ x] Operating system: Debian 12

augustinust, 2024-04-24:
###  update
I started a *open62541*-server with a method returning *uncertain*.
Then I called the method with *UaExpert* - and I get a result.

But with the *open62541*-client code, the result is empty.

augustinust, 2024-04-25:
I fixed the issue locally by editing the method *UA_Client_call* in the file *ua_client_highlevel.c*.

Original:
```
    if(retval == UA_STATUSCODE_GOOD) {
        if(response.resultsSize == 1)
            retval = response.results[0].statusCode;
        else
            retval = UA_STATUSCODE_BADUNEXPECTEDERROR;
    }
    if(retval != UA_STATUSCODE_GOOD) {
        UA_CallResponse_clear(&response);
        return retval;
    }
```
Slightly modified version:

```
    if(retval == UA_STATUSCODE_GOOD || retval == UA_STATUSCODE_UNCERTAIN) {
        if(response.resultsSize == 1)
            retval = response.results[0].statusCode;
        else
            retval = UA_STATUSCODE_BADUNEXPECTEDERROR;
    } else {
        UA_CallResponse_clear(&response);
        return retval;
    }
```

tboeckel, 2024-04-26:
I also encountered the problem that none of the output parameters are returned to the calling client if a method returns a different status code than UA_STATUSCODE_GOOD.
Therefore all methods of my servers always return the status code UA_STATUSCODE_GOOD, because calling the method from an OPC-UA point of view was successful.
Additionally all methods always have at least two output parameters (success:bool and message:string). If everything went alright, then success=true and message="ok" will be returned, otherwise success=false and message="whatever went wrong" will be returned. That way it is easy to check wether the method call was successful at all and to return arbitrarily verbose informations about possible failures.

augustinust, 2024-04-26:
The server sends values when the *status code* is *uncertain*, but the *open62541*-client does not return them.
When I use another client(UaExpert, node-opcua), I get a result.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier