open62541/open62541 #6460: UA_Client_connect() is blocking

spole340, 2024-05-07:
## Description
I tried to run example client_connect_loop.c, but the function UA_Client_connect() is blocking when OPC server is unreachable.
The program stuck here and if server is reachable it is not reconnected. I want to use UA_Client_connectUsername() insteat of UA_Client_connect(), but behavior is same (blocking). I need non blocking function that returns me something if something wrong. Can someone help me, please?

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Operating system: Ubuntu 22.04

jpfr, 2024-05-19:
Are you using the 1.4 version?

we have dedicated _async versions for the connect methods. Use those.

It can take some time for the connection to fail. That depends on the DNS timeout configured in your operating system.

Schokoladenblau, 2024-06-20:
I got the same error. The log shows the folling:
[2024-06-20 12:48:11.789 (UTC+0000)] info/eventloop	Starting the EventLoop
[2024-06-20 12:48:11.789 (UTC+0000)] warn/client	skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-06-20 12:48:11.789 (UTC+0000)] info/network	TCP 8	| Opening a connection to "192.168.0.1" on port 4840
[2024-06-20 12:48:13.848 (UTC+0000)] info/network	TCP 8	| The connection closes with error 111
[2024-06-20 12:48:13.848 (UTC+0000)] info/channel	TCP 8	| SC 0	| SecureChannel closed
[2024-06-20 12:48:13.848 (UTC+0000)] error/client	Could not open a TCP connection to the Endpoint at opc.tcp://192.168.0.1:4840
[2024-06-20 12:48:13.848 (UTC+0000)] info/client	Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionRejected
[2024-06-20 12:48:13.848 (UTC+0000)] info/network	TCP 8	| Socket closed

But after that the function doas not return, it gets stuck. 
Git commit hash is 3d0da9ad73b49058e197913352bb0eb30235c14b
I am using it inside Docker with the base image mcr.microsoft.com/devcontainers/cpp:1-debian-11

Note: If the Server is reachable everything works fine.

hdabis1234, 2024-09-20:
I got the same from a linux platform. Works fine in Windows. Any updates on this?

BTW: About reconnections:  have you tried setting noReconnect to true to stop it from reconnecting.

closed 2026-10-20 by o6-nbeier