open62541/open62541 #6461: Potential system resource/memory leaks

10066847, 2024-05-08:
## detected at 1.3.9 or before

## socket handle failed to close

**line 1367, file ua_client_connect.c,** the call to UA_SecureChannel_close might miss the call to close the socket handle when the SecureChannel was not attached. I added the following codes before the call seems to be able to work around this issue.
...
**if(!client->channel.connection) {
        if(client->connection.close) 
            client->connection.close(&client->connection);
}**
UA_SecureChannel_close(&client->channel);
...

## potenial memory leak

**line 118, file ua_accesscontrol_default.c**, I detected the 'username' not always be clear. It seems to be caused by reconnect, I will enclose more details soon, it hard to reproduce, it sometimes takes days or weeks.
UA_ByteString *username = UA_ByteString_new();
if(username)
    UA_ByteString_copy(&userToken->userName, username);
  ***sessionContext = username;**

## dangled logger
**line 41, file ua_client_connect.c**, it seems to that the logger requires to update or it will cause memory access violation.

static UA_SecurityPolicy *
getSecurityPolicy(UA_Client *client, UA_String policyUri) {
    for(size_t i = 0; i < client->config.securityPoliciesSize; i++) {
        if(UA_String_equal(&policyUri, &client->config.securityPolicies[i].policyUri)) {
            UA_SecurityPolicy *sp = &client->config.securityPolicies[i];
            **sp->logger = &client->config.logger;**
            return sp;
        }
    }
    return NULL;
}

Used CMake options:

```bash
cmake  -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENCRYPTION_PLUGINS=OPENSSL -DUA_ENABLE_DIAGNOSTICS=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_DELTAFRAMES=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=ON
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-05-08:
Thanks for the report.
We will look into that and come back with patch releases where required.

10066847, 2024-05-09:
to follow up ##potenial memory leak
The log print is enclosed as followed:
[2024-05-08 14:48:22.398 (UTC+0800)] info/channel	Connection -1 | SecureChannel 1 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 14:55:52.409 (UTC+0800)] info/channel	Connection -1 | SecureChannel 1 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 15:03:22.429 (UTC+0800)] info/channel	Connection -1 | SecureChannel 1 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 15:10:52.431 (UTC+0800)] info/channel	Connection -1 | SecureChannel 1 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 15:18:22.461 (UTC+0800)] info/channel	Connection -1 | SecureChannel 1 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 15:25:52.467 (UTC+0800)] info/channel	Connection -1 | SecureChannel 1 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 15:30:12.789 (UTC+0800)] info/securitypolicy	The basic256sha256 security policy channel with openssl is created.
[2024-05-08 15:30:12.814 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 600.00s
**[2024-05-08 15:30:13.187 (UTC+0800)] info/securitypolicy	The Aes128Sha256RsaOaep security policy channel with openssl is created.
[2024-05-08 15:30:13.197 (UTC+0800)] info/securitypolicy	The Aes128Sha256RsaOaep security policy channel with openssl is deleted.
[2024-05-08 15:30:13.197 (UTC+0800)] info/session	SecureChannel 2 | Session "urn:DESKTOP-USUFLG6:UnifiedAutomation:UaExpert" | ActivateSession: Session attached to new channel
[2024-05-08 15:30:13.197 (UTC+0800)] info/session	SecureChannel 2 | Session "urn:DESKTOP-USUFLG6:UnifiedAutomation:UaExpert" | ActivateSession: Session activated**
[2024-05-08 15:30:17.961 (UTC+0800)] info/securitypolicy	The basic256sha256 security policy channel with openssl is deleted.
[2024-05-08 15:37:42.855 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 15:45:12.878 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 15:52:42.902 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 16:00:12.916 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 16:07:42.974 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 16:15:12.947 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 600.00s
[2024-05-08 16:22:42.943 (UTC+0800)] info/channel	Connection -1 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 600.00s

The same time log print from UaExpert as followed:
**15:30:11.005 | Server Node        | OPC UA Application             | Connection status of server 'OPC UA Application' changed to 'ConnectionErrorApiReconnect'.
15:30:16.198 | Server Node        | OPC UA Application             | Connection status of server 'OPC UA Application' changed to 'Connected'.**

The log in bold memory leak happens.

teccheck, 2025-01-09:
UA_SecureChannel seems to have changed quite a lot with since 1.3.9 so I marked this issue as revision needed. But this is likely not a problem anymore

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier