open62541/open62541 #6463: Restart of server fails because of disabled reuseaddr socket option

keba-estr, 2024-05-08:
## Description
As the socket option reuseaddr is disabled by default, a socket to the same address can only be opened after a system specific timeout (e.g. 1 minute in Linux, 2 minutes in Windows).
The server opens a listening connection when the event loop initialized. If this connection fails, there is no retry.
If the server is closed and restarted within the above timeout, opening the listing connection fails.

The event loop should retry opening the listing connection within the system specific socket timeout.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4
 - [X] Operating system: Linux
 - [ ] Critical issue

jpfr, 2024-05-09:
The fix is here: https://github.com/open62541/open62541/pull/6466

closed 2024-05-14 by jpfr

keba-estr, 2024-05-23:
Problem not resolved, error still happens.