open62541/open62541 #6480: Use UA_ClientConfig_setDefaultEncryption

Malick31, 2024-05-15:
## Description
I have an issue with use UA_ClientConfig_setDefaultEncryption, I dont know if its still usable. When I try to use it I have
have an undifined functions, but its in the same file as UA_ClientConfig_setDefault which I can use, d I need to do something specil before I use it? 
## Background Information / Reproduction Steps
I use it for encryption because I tried to connect to bnr with a username and a password

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

keba-estr, 2024-06-14:
You have to set the define UA_ENABLE_ENCRYPTION to activate the function UA_ClientConfig_setDefaultEncryption.

teccheck, 2025-01-09:
As far as I can see, the answer above should solve your problem. If not, feel free to reopen this issue

closed 2025-01-09 by teccheck