open62541/open62541 #6481: Path not found in open62541Config.cmake

keba-estr, 2024-05-16:
## Description
Due to commit 6091b1865f87acf3285b45b06c38d740cab40cd5 the path usr/share/open62541/tools/ua-nodeset is not available any more. Yet the cmake file open62541Config.cmake checks this path.
Therefore the compilation of this cmake file fails.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4.1
 - [X] Critical issue

keba-estr, 2024-05-16:
Line in open62541Config.cmake:
set_and_check(UA_NODESET_DIR @PACKAGE_open62541_install_nodeset_dir@ CACHE PATH "Path to the directory that contains the OPC UA schema repository")

@jpfr: What is the correct solution? Should the check be deleted or should the path be replaced?

jpfr, 2024-05-16:
You are right.
The reduced nodeset (used to build the library) is in usr/share/open62541/schema.

keba-estr, 2024-05-16:
@jpfr Could you please create a pull request? I don't understand how '@PACKAGE_open62541_install_nodeset_dir@' is resolved or which would be the correct path variable here.

keba-estr, 2024-05-17:
I will try out usr/share/open62541/schema in CMakeLists.txt to see if other errors occur.

keba-estr, 2024-05-17:
For the compile option UA_NAMESPACE_ZERO = "FULL" the file ${UA_NODESET_DIR}/Schema/Opc.Ua.NodeSet2.xml is still referenced in CMakeLists.txt and open62541Macros.cmake, but not installed to the tools package any more.

keba-estr, 2024-05-17:
Created PR #6483

keba-estr, 2024-05-17:
Resolved with PR #6483

closed 2024-05-17 by keba-estr