open62541/open62541 #6485: Missing file libopen62541.so for debian linking

keba-estr, 2024-05-17:
## Description
The file lobopen62541.so has been removed from the libopen62541-dev package with commit 9e0b766fbba73be8d703b6ccbfdeec5bfd200b48.
The argument was that the so file is already available in the libopen62541 package.
But the name of the so file in the libopen62541 package is libopen62541.so.1, and the linker only looks for files with the extension .so.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4
 - [X] Operating system: Linux (debian)
 - [X] Critical issue

keba-estr, 2024-05-17:
@jpfr: What is the correct solution?
- Add the so file to the dev package again
- Add the so file to the libopen62541 package by extending the install-template with usr/lib/*/*.so

keba-estr, 2024-05-17:
Created PR #6486

jpfr, 2024-05-19:
The Debian package generated on my box generates both the .so.1 and also a symbolic link to the .so.

Maybe you generate packages differently.
I use ./tools/prepare_packaging.py and then ‘debuild’ from the source root folder.

keba-estr, 2024-05-21:
> The Debian package generated on my box generates both the .so.1 and also a symbolic link to the .so.
> 
> Maybe you generate packages differently. I use ./tools/prepare_packaging.py and then ‘debuild’ from the source root folder.

I also use ./tools/prepare_packaging.py, but then I use a KEBA specific function 'buildpackage'.
So I will have to handle the topic within my KEBA specific build part.

closed 2024-05-21 by keba-estr