open62541/open62541 #6487: Add version resource to Windows shared library DLL

ahuj9, 2024-05-18:
## Description

Open62541 when built as a shared library on Windows could use a version resource .rc file configured from ${OPEN62541_VER_X} CMake variables.

jpfr, 2024-05-19:
Sounds good. Where should this be installed?

irostagno, 2024-05-20:
The easy way is to add [this](https://github.com/halex2005/CMakeHelpers/blob/master/generate_product_version.cmake), [this](https://github.com/halex2005/CMakeHelpers/blob/master/VersionInfo.in) and [this](https://github.com/halex2005/CMakeHelpers/blob/master/VersionResource.rc) to `tools/cmake`. Then in the `CMakeLists.txt` you should add:
```
generate_product_version(
    open62541_resource_version
    NAME "open62541"
    VERSION_MAJOR ${OPEN62541_VER_MAJOR}
    VERSION_MINOR $OPEN62541_VER_MINOR}
    VERSION_PATCH ${OPEN62541_VER_PATCH}
    VERSION_REVISION ${OPEN62541_VER_LABEL}
)
```

This generates variable `open62541_resource_version` to add into the `add_library` command:
```
add_library(open62541 ... ${open62541_resource_version})
```

For Windows, metadata is embedded into the DLL, so, no installation is needed.

OgreTransporter, 2024-05-25:
It is not necessary to make it so complicated with macros and different files. A simple template and `configure_file` is completely sufficient.