open62541/open62541 #6489: badtokenidentity

Malick31, 2024-05-21:
## Description
I have an issue while usinf UA_client_connect_Username, I did follow a lot of thing, I have enable encryption, and I have
installer mbedtls, i've used de python scritp to generate .der file, and then i've follow the client_ecnryptio.c file but I still have some issus with bad token_identity 
## Background Information / Reproduction Steps

Used CMake options:

cmake -DUA_ENABLE_ENCRYPTION=ON -DUA_BUILD_EXAMPLES=OFF -DUA_BUILD_SELFSIGNED_CERTIFICATE=ON -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..

Include all CMake options here, which you modified or used for your build.

If you are using cmake-gui, go to "Tools > Show my Changes" and paste the content of "Command Line Options"

On the command line use `cmake -L` (or `cmake -LA` if you changed advanced variables)
-->

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

### this is my code 

#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>
#include <open62541/plugin/log_stdout.h>
#include <open62541/plugin/securitypolicy.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <stdlib.h>
#include <iostream>
#include "common.h"

int main(void) {
    std::cout << "bonjour" << std::endl;
    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    std::cout << "config" << std::endl;

    UA_ByteString cert = loadFile("server_cert.der");
    std::cout << "key" << std::endl;
    UA_ByteString key = loadFile("server_key.der");

    size_t trustListSize = 0;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize);

    UA_ByteString* revocationList = NULL;
    size_t revocationListSize = 0;

    std::cout << "before setdefaultconfig" << std::endl;

    UA_StatusCode ret = UA_ClientConfig_setDefaultEncryption(cc, cert, key,
        trustList, trustListSize,
        revocationList, revocationListSize);
    if(ret != UA_STATUSCODE_GOOD) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                    "Failed to set encryption." );
        UA_Client_delete(client);
        return EXIT_FAILURE;
    }
    std::cout << "setdefaultconfig" << std::endl;

    UA_StatusCode retval = UA_Client_connectUsername(client, "opc.tcp://127.0.0.1:4841", "ab", "ab");
    if(retval != UA_STATUSCODE_GOOD)
    {
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                    "The connection failed with status code %s",
                    UA_StatusCode_name(retval));
        UA_Client_delete(client);
        return 0;
    }
    std::cout << "error" << std::endl;

    UA_Client_disconnect(client);
    UA_Client_delete(client);
    UA_ClientConfig_clear(cc);

    return 0;
}

##### the error
[2024-05-21 16:25:00.308 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Basic128Rsa15 with error code BadInvalidArgument
[2024-05-21 16:25:00.308 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Basic256 with error code BadInvalidArgument
[2024-05-21 16:25:00.310 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Aes256Sha256RsaPss with error code BadInvalidArgument
[2024-05-21 16:25:00.310 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Basic256Sha256 with error code BadInvalidArgument
[2024-05-21 16:25:00.312 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Aes128Sha256RsaOaep with error code BadInvalidArgument
setdefaultconfig
[2024-05-21 16:25:00.313 (UTC+0200)] info/eventloop     Starting the EventLoop
[2024-05-21 16:25:00.313 (UTC+0200)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-05-21 16:25:00.317 (UTC+0200)] info/network       TCP 324 | New connection to "127.0.0.1" on port 4841
[2024-05-21 16:25:00.317 (UTC+0200)] info/channel       TCP 324 | SC 2724020538 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-05-21 16:25:00.318 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-05-21 16:25:00.318 (UTC+0200)] info/client        The initially defined EndpointURL opc.tcp://127.0.0.1:4841 is valid for the server
[2024-05-21 16:25:00.319 (UTC+0200)] info/client        Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
[2024-05-21 16:25:00.319 (UTC+0200)] info/client        Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256' not available
[2024-05-21 16:25:00.319 (UTC+0200)] info/client        Rejecting UserTokenPolicy 2 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2024-05-21 16:25:00.320 (UTC+0200)] info/client        Rejecting UserTokenPolicy 3 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2024-05-21 16:25:00.321 (UTC+0200)] info/client        Rejecting UserTokenPolicy 4 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256' not available
[2024-05-21 16:25:00.321 (UTC+0200)] info/client        Rejecting endpoint 1: security policy not available
[2024-05-21 16:25:00.322 (UTC+0200)] info/client        Rejecting endpoint 2: security policy not available
[2024-05-21 16:25:00.322 (UTC+0200)] info/client        Rejecting endpoint 3: security policy not available
[2024-05-21 16:25:00.322 (UTC+0200)] info/client        Rejecting endpoint 4: security policy not available
[2024-05-21 16:25:00.323 (UTC+0200)] info/client        Rejecting endpoint 5: security policy not available
[2024-05-21 16:25:00.324 (UTC+0200)] info/client        Rejecting endpoint 6: security policy not available
[2024-05-21 16:25:00.324 (UTC+0200)] error/client       No suitable UserTokenPolicy found for the possible endpoints
[2024-05-21 16:25:00.324 (UTC+0200)] info/client        Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadIdentityTokenInvalid
[2024-05-21 16:25:00.328 (UTC+0200)] info/userland      The connection failed with status code BadIdentityTokenInvalid
[2024-05-21 16:25:00.328 (UTC+0200)] info/channel       TCP 324 | SC 2724020538 | SecureChannel closed
[2024-05-21 16:25:00.329 (UTC+0200)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadIdentityTokenInvalid
[2024-05-21 16:25:00.329 (UTC+0200)] info/network       TCP 324 | Socket closed
[2024-05-21 16:25:00.329 (UTC+0200)] info/eventloop     Stopping the EventLoop
[2024-05-21 16:25:00.330 (UTC+0200)] info/network       UDP     | Shutting down the ConnectionManager
[2024-05-21 16:25:00.330 (UTC+0200)] info/network       TCP     | Shutting down the ConnectionManager
[2024-05-21 16:25:00.330 (UTC+0200)] info/eventloop     The EventLoop has stopped

closed 2024-05-23 by Malick31

aaghashr, 2024-11-21:
@Malick31 

I'm encountering the same problem. Have you managed to find a solution? Would love to hear your insights!

Malick31, 2024-11-22:
@aaghashr  
Finally, I used OpenSSL. I think it's easier. I installed it, added it to the environment variables, and then simply ran the following command:
cmake -DUA_ENABLE_ENCRYPTION=OPENSSL -DBUILD_SHARED_LIBS=ON ..
And that was it!