open62541/open62541 #6493: Restart of server fails because of disabled reuseaddr socket option

keba-estr, 2024-05-23:
## Description
Follow up of issue #6463 

## Background Information / Reproduction Steps
The provided fix with commit 00206cdffa0d90c1d3d423eb4ecea7c997eccda2 does not work (at least on linux systems).
Although the function UA_EventLoopPOSIX_setReusable succeeds with UA_STATUSCODE_GOOD when closing the port,
the socket can not be opened sporadically at the next restart.
It seems that the socket option SO_REUSEADDR only has an effect if called before the function bind.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4
 - [X] Operating system: Linux

jpfr, 2024-05-23:
In that case you best option is to set the reuseaddr configuration flag for your application.

https://github.com/open62541/open62541/blob/9229a2fabef5137940f9481866eea44434c9a350/include/open62541/server.h#L181

That way the socket is set reusable automatically.

keba-estr, 2024-05-23:
We can set the option in the server.
@jpfr: However, the security point of view about this setting should be discussed in the next community call.

jpfr, 2024-05-23:
Sure we can discuss.

closed 2024-05-23 by jpfr