open62541/open62541 #6502: use open62541 on esp32

Ivan0x17, 2024-05-24:
i am using vscode and platformio on windows 11

I would like to be able to implement the library in my project with an esp32 how can I do , I admit I am new in this world and I would thank you very much if someone can explain me step by step how to do it , the project is already created , but I don't know how to move and what to do

does this link have anything to do with it?
https://www.open62541.org/doc/master/building.html

I have already cloned the library with git

r-sin, 2024-05-29:
Might be worth having a look at https://github.com/Pro/open62541-esp32

Ivan0x17, 2024-05-29:
> Might be worth having a look at

Ok i Look a github project

Ivan0x17, 2024-05-29:
> Might be worth having a look at https://github.com/Pro/open62541-esp32
> 
> 

In this project:
https://github.com/Pro/open62541-esp32

The developer use a idf espressif not platformio