open62541/open62541 #6508: Case mismatch for 'Schema' directories

keba-estr, 2024-05-27:
## Description
Nodeset generation might fail on case dependent Linux systems, as the directory referenced in the file open62541Macros.cmake is '${UA_NODESET_DIR}/Schema', but in the debian tools package the folder name is 'schema' with lower case letters.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4.1
 - [X] Operating system: Linux

keba-estr, 2024-05-27:
@jpfr: What is the correct solution here?
It seems that UA_NODESET_DIR is both used with 'open62541/deps/ua-nodeset' for the open62541 internal nodeset generation and with '@PACKAGE_open62541_install_nodeset_dir@' for a user nodeset generation.
Should the case of the 'schema' directory in the debian package be the same as in the dependent nodesetLoader component?

jpfr, 2024-05-27:
The open62541Config.cmake contains this line:

set_and_check(UA_NODESET_DIR ${PACKAGE_PREFIX_DIR}/share/open62541/schema CACHE PATH "Path to the directory that contains the schema files")

In our internal CMakeLists.txt we have both nodeset-dir and schema-dir:

set(UA_SCHEMA_DIR ${PROJECT_SOURCE_DIR}/tools/schema) # Directory with the schema files for installation

My proposal is to unify that and remove UA_SCHEMA_DIR.
So that the UA_NODESET_DIR points to wherever the schemas are.
Including either /schema or /Schema.

keba-estr, 2024-05-31:
@jpfr: Added possible solution with PR #6519, but I am not sure if I got everything correct.

PR #6519 (2024-06-02): [REVIEW] fix(build): Correct path for schema files of external build
Use UA_SCHEMA_DIR for path to schema file to be independent from the different case of the 'schema' and 'Schema' folder.
Resolves #6508
closed 2024-06-03 by keba-estr