open62541/open62541 #6510: Open62541 Client implementation to monitor 1000 variables on a server

r1gbymordeca1, 2024-05-27:
Client implementation to monitor 1000 variables on a server #6509

I am working on a client implementation where I need to subscribe to 1000 variables from a server. Right now, I am creating one subscription and adding the monitored variables using UA_Client_MonitoredItems_createDataChange() to the same subscription. Also, I am handling all the datachange callbacks through the same callback mechanism.
 
Is this approach good or should I create multiple subscriptions and add monitor items to them accordingly?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): HEAD
 - [ ] Other OPC UA SDKs used (client or server): No
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: No
 - [ ] Wireshark network dump attached: No
 - [ ] Self-contained code example attached: No
 - [x] Critical issue: YES

jpfr, 2024-05-27:
Using just one subscription is preferred.
If you want to receive all notifications together..

jpfr, 2024-05-27:
I guess this is *not* a critical issue then.

closed 2024-05-27 by jpfr