open62541/open62541 #6512: Monitored Items and UA_VARIANT_DATA_NODELETE

basyskom-jvoe, 2024-05-28:
When a data source read callback sets the storage type of the `UA_Variant`  to ` UA_VARIANT_DATA_NODELETE` and the data pointer refers to a fixed memory position (e.g. static variable), any monitored item for a variable node with such a data source callback doesn't detect any data changes.

If I remember correctly, there were changes to the behavior of variant copying with `UA_VARIANT_DATA_NODELETE` some time ago. Would these changes explain the observed behavior and if so, is there a way to deal with it?

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.4.1

jpfr, 2024-05-30:
That is strange.

what could happen is that we keep the original variant around for the comparison.
So when we compare the old variant still points to the static memory that now contains the new value.

jpfr, 2024-05-30:
Added a unit test for this in #6518.
It appears to work correctly.

basyskom-jvoe, 2024-05-31:
This is indeed very strange...

I have added debug output to `detectValueChange()`:
```
    UA_LOG_INFO_SUBSCRIPTION(server->config.logging, mon->subscription, "XXXXXXXXXXXXXXXXXXXXXX Monitored item %p vs. %p", mon->lastValue.value.data, dv->value.data);
```
and according to this output, the data pointer is indeed identical for the old and new variants:
```
Session "urn:rigel:UnifiedAutomation:UaExpert"	| Subscription 3 | XXXXXXXXXXXXXXXXXXXXXX Monitored item 0xdab908 vs. 0xdab908
```

jpfr, 2024-06-01:
_NODELETE can still be used for zero-copy reads.
Only MonitoredItems we persist the value so we are able to compare with the next sample.

But you make a good point.
We could check if the value is pointer-free (e.g. a number) and then do a memcpy over the already allocated previous memory...

basyskom-jvoe, 2024-06-03:
@jpfr If I understand the current implementation correctly, a deep copy is always made directly after the data source read callback is called.
What you describe is how it remember the implementation in 0.3 (only the initial DataValue and any changed DataValue is deep copied, the comparison is done with the "original" without copying it first).

jpfr, 2024-06-03:
Yeah. We had issues when keeping around _NODELETE variables after releasing the node in the nodestore.
So now we make a deep copy (if not already the case) internally.

What we can do is: Delay the node _release until after the MonitoredItem is fully processed.
For that we need some kind of global list (in the server struct) with to-be-released nodes.

udovichenkoAlexander, 2026-01-16:
So now zero-copy not possible ?