open62541/open62541 #6520: Can no longer write UA_Int32 to an enumeration node

rickboks, 2024-05-31:
Hi! I have been on `open62541` version `1.3.6` for about a year, and tried to switch to version `1.4.1` today. In my application, I dynamically define enumeration types as outlined in https://github.com/open62541/open62541/issues/2032#issuecomment-474752649 . This always worked perfectly in `v1.3.6`. The type definition also works fine in `v1.4.1`, but when I try to create a node with the created enum type using `UA_Server_addVariableNode`, I get an error `BadTypeMismatch` and the following log message:
```
[2024-05-31 17:28:23.103 (UTC+0200)] warning/server	Writing the value of Node ns=2;s=myEnumInstance failed with the following reason: DataType of the value is incompatible
```
I found out that in commit 0d0b18b08bd27d5e7da17dd40bc115b541bb7993, a special case for accepting `UA_Int32` writes to Enumerations was removed from the function `compatibleDataTypes`. This is the cause of my problem; if I add the check to `open62541` version `v1.4.1`, my enums work again. Why was this special case removed? Can it be re-introduced, or do I need to instantiate enumeration variables differently?

The relevant check that was removed from `compatibleDataTypes` is:

```
  /* The constraint is an enum -> allow writing Int32 */
  if(UA_NodeId_equal(dataType, &UA_TYPES[UA_TYPES_INT32].typeId) &&
     isNodeInTree_singleRef(server, constraintDataType, &enumNodeId,
                            UA_REFERENCETYPEINDEX_HASSUBTYPE))
      return true;
```
 Thanks in advance!

jpfr, 2024-06-01:
@andreasebner Please check whether your commit can be reverted.
In any case this needs documentation and better log messages.

jpfr, 2024-06-04:
According to the unit tests this is working:
https://github.com/open62541/open62541/pull/6532/commits/e87db6c07116595dfdbf1eddecd8ec11cb4dba12

If you please extend this unit test to reproduce the problem.
Then we can fix it easily.

rickboks, 2024-06-05:
Thanks @jpfr. I think the difference is that the test operates on a predefined enum `UA_TYPES_MESSAGESECURITYMODE`, while I would like it to work on dynamically created 'custom' enums. I created a pull request #6536 to include such a custom enum in the tests. Currently the test fails. I checked that the test succeeds again if I re-introduce the check which was removed in 0d0b18b

jpfr, 2024-06-05:
There should be no difference between built in and custom types. But ok. Will check.

rickboks, 2024-06-06:
@thdieterle see the comments in #6536 for the solution

closed 2024-06-06 by rickboks