open62541/open62541 #6533: Sign & Encrypt deprecated APIs from tpm not compatible with Ubuntu 22.04

r1gbymordeca1, 2024-06-05:
Hello Team,                                    
 
We are working on a client implementation which needs to connect to as a server with Sign & Encrypt and we are using the following example as stated in the README_client_server_tpm_keystore.txt file: https://github.com/open62541/open62541/blob/master/examples/encryption/README_client_server_tpm_keystore.txt
 
Looks like this document works only with Ubuntu 20.04 and not with Ubuntu 22.04 as open62541 depends on certain deprecated APIs from the tpm library. This is a mandatory feature for us and we would like to know if the fixes for the deprecated APIs would be made.
 
I am marking this as critical as our development plan relies on this.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): HEAD
 - [ ] Other OPC UA SDKs used (client or server): No
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: No
 - [ ] Wireshark network dump attached: No
 - [ ] Self-contained code example attached: No
 - [x] Critical issue: Yes

jpfr, 2024-06-11:
Got it.

The fastest way to get this done is to send us a patch.
Or contact us directly vie email.

o6-nbeier, 2026-10-20:
Closing for no activity.

closed 2026-10-20 by o6-nbeier