open62541/open62541 #6558: UA_ClientConfig_setDefaultEncryption - unresolved external symbol - UaCPPServer - UA Expert - Unreal Engine 5.3.2

Naosukii, 2024-06-21:
## Description

Hello,

I want to create a Plugin for the Unreal 5.3.2 to connect with a OPC UA Server.

The Plugin is already connecting to my UaCPPServer (Anonymous and Security: None), it also can read the node values with "UA_Client_connect" in runtime.

But I want to disable "Anonymous" and use Basic256Sha256 Encryption.
I have set up the UaCPPServer and the UaExpert to get a connection with username/password.

In the c++ I try to call "UA_ClientConfig_setDefaultEncryption" but get following error:

`Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	LNK2019	unresolved external symbol __imp_UA_ClientConfig_setDefaultEncryption referenced in function "public: static bool __cdecl UOpcUaClientBPLibrary::UA_Client_connectUsername_open62541(class FString,class FString,class FString)" ...		
`

This is my code inside my .cpp

```c++
UA_ByteString* loadFile(const char* path)
{
	FILE* f = fopen(path, "rb");
	if (f == NULL)
		return NULL;

	fseek(f, 0, SEEK_END);
	long fsize = ftell(f);
	fseek(f, 0, SEEK_SET);  /* same as rewind(f); */

	UA_ByteString* result = UA_ByteString_new();
	UA_ByteString_allocBuffer(result, (size_t)fsize + 1);
	memset(result->data, 0, result->length);
	fread(result->data, result->length, 1, f);
	fclose(f);

	return result;
}

bool UOpcUaClientBPLibrary::UA_Client_connectUsername_open62541(FString endpointUrl, FString username, FString password)
{
	// Load certificate generated by UA-Expert
	UA_ByteString* certificate = loadFile("uaexpert.der");
	UA_ByteString* privateKey = loadFile("uaexpert_key.pem");

	// Trust list;
	size_t trustListSize = 0;
	UA_STACKARRAY(UA_ByteString, trustList, trustListSize);

	// Revocation list
	UA_ByteString* revocationList = NULL;
	size_t revocationListSize = 0;

	// Config
	client = UA_Client_new();
	UA_ClientConfig* config = UA_Client_getConfig(client);
	config->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
	UA_ClientConfig_setDefaultEncryption(config, *certificate, *privateKey, trustList, trustListSize, revocationList, revocationListSize);

	// Set description as specified in certificate
	config->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("de", "UaExpert");
	config->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;
	config->clientDescription.applicationUri = UA_STRING_ALLOC("urn:DESKTOP-9NC4I2B:UnifiedAutomation:UaExpert");

	// Connect
	UA_StatusCode retval = UA_Client_connectUsername(client, TCHAR_TO_UTF8(*endpointUrl), TCHAR_TO_UTF8(*username), TCHAR_TO_UTF8(*password));
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return false;
	}

	return true;
}
``` 

Sorry I'm really new into github and opcua with c++ / unreal.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): UaCPP Server // UaExpert
 - [ ] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

cxjzl, 2024-07-23:
Hey buddy, have you solved your problem? I have the same problem as you

Naosukii, 2024-07-23:
> Hey buddy, have you solved your problem? I have the same problem as you

Yes, I had to build a new .dll / .lib with cmake. I used the cmake-gui.

o6-nbeier, 2026-10-20:
Closing for now.

closed 2026-10-20 by o6-nbeier