open62541/open62541 #6562: When the Client connection fails, the Session on the Server is not closed.

yirongliang, 2024-06-26:
## Description
I use UA.NET's Quickstarts.ReferenceServer as the test server and open62541 as the client test.

I'm testing that the Server returns the BadIdentityTokenRejected error status.
When the connection has been closed, the Session is still retained on the Server side. This causes every time I re-establish a connection, the Server side will continue to generate new Sessions until the Server can no longer accept new connections.
I used UaExpert to conduct the same BadIdentityTokenRejected test, and it can close the Server-side Session normally.

The following is the message from my single connection:

[2024-06-26 17:00:23.680 (UTC+0800)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2024-06-26 17:00:23.681 (UTC+0800)] info/securitypolicy        The basic256 security policy with openssl is added.
[2024-06-26 17:00:23.681 (UTC+0800)] info/securitypolicy        The Aes256Sha256RsaPss security policy with openssl is added.
[2024-06-26 17:00:23.682 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2024-06-26 17:00:23.682 (UTC+0800)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2024-06-26 17:00:23.683 (UTC+0800)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2024-06-26 17:00:23.683 (UTC+0800)] info/securitypolicy        The basic256 security policy with openssl is added.
[2024-06-26 17:00:23.684 (UTC+0800)] info/securitypolicy        The Aes256Sha256RsaPss security policy with openssl is added.
[2024-06-26 17:00:23.685 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2024-06-26 17:00:23.685 (UTC+0800)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2024-06-26 17:00:23.687 (UTC+0800)] info/eventloop     Starting the EventLoop
[2024-06-26 17:00:23.689 (UTC+0800)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-06-26 17:00:23.691 (UTC+0800)] info/network       TCP 1688        | Opening a connection to "allen-icpdas" on port 62541
[2024-06-26 17:00:23.693 (UTC+0800)] info/channel       TCP 1688        | SC 11 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-06-26 17:00:23.695 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-06-26 17:00:23.701 (UTC+0800)] info/client        The initially defined EndpointURL opc.tcp://allen-icpdas:62541/Quickstarts/ReferenceServer is valid for the server
[2024-06-26 17:00:23.704 (UTC+0800)] warn/client        The server returned Endpoints with a different EndpointUrl opc.https://allen-icpdas:62540/Quickstarts/ReferenceServer than was used to initialize the connection: opc.tcp://allen-icpdas:62541/Quickstarts/ReferenceServer. Some servers require a complete match of the EndpointUrl/DiscoveryUrl (including the path) to return all endpoints.
[2024-06-26 17:00:23.707 (UTC+0800)] info/client        Rejecting endpoint 1: security mode doesn't match
[2024-06-26 17:00:23.710 (UTC+0800)] info/client        Rejecting endpoint 2: security mode doesn't match
[2024-06-26 17:00:23.713 (UTC+0800)] info/client        Rejecting endpoint 3: security mode doesn't match
[2024-06-26 17:00:23.714 (UTC+0800)] info/client        Rejecting endpoint 4: security mode doesn't match
[2024-06-26 17:00:23.715 (UTC+0800)] info/client        Rejecting UserTokenPolicy 0 (anonymous) in endpoint 5: configuration doesn't match
[2024-06-26 17:00:23.718 (UTC+0800)] info/client        Rejecting UserTokenPolicy 1 (username) in endpoint 5: configuration doesn't match
[2024-06-26 17:00:23.720 (UTC+0800)] info/client        Rejecting endpoint 6: security policy doesn't match.
[2024-06-26 17:00:23.720 (UTC+0800)] info/client        Rejecting endpoint 7: security policy doesn't match.
[2024-06-26 17:00:23.721 (UTC+0800)] info/client        Selected endpoint 5 in URL opc.tcp://allen-icpdas:62541/Quickstarts/ReferenceServer with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2024-06-26 17:00:23.722 (UTC+0800)] info/client        Selected UserTokenPolicy 15 with UserTokenType Certificate and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2024-06-26 17:00:23.723 (UTC+0800)] info/channel       TCP 1688        | SC 11 | SecureChannel closed
[2024-06-26 17:00:23.724 (UTC+0800)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-06-26 17:00:23.725 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy channel with openssl is created.
[2024-06-26 17:00:23.728 (UTC+0800)] info/network       TCP 1716        | Opening a connection to "allen-icpdas" on port 62541
[2024-06-26 17:00:23.731 (UTC+0800)] info/network       TCP 1688        | Socket closed
[2024-06-26 17:00:23.758 (UTC+0800)] warn/userland      No certificate store configured. Accepting the certificate.
[2024-06-26 17:00:23.772 (UTC+0800)] info/channel       TCP 1716        | SC 12 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 600.00s
[2024-06-26 17:00:23.773 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-06-26 17:00:23.789 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy channel with openssl is created.
[2024-06-26 17:00:23.796 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy channel with openssl is deleted.
[2024-06-26 17:00:23.796 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2024-06-26 17:00:23.806 (UTC+0800)] info/client        Received a ServiceFault response
[2024-06-26 17:00:23.806 (UTC+0800)] error/client       Session cannot be activated with StatusCode BadIdentityTokenRejected. The client cannot recover from this, closing the connection.
[2024-06-26 17:00:23.808 (UTC+0800)] info/client        Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
[2024-06-26 17:00:23.811 (UTC+0800)] info/channel       TCP 1716        | SC 12 | SecureChannel closed
[2024-06-26 17:00:23.812 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy channel with openssl is deleted.
[2024-06-26 17:00:23.813 (UTC+0800)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
[2024-06-26 17:00:23.814 (UTC+0800)] info/network       TCP 1716        | Socket closed
[2024-06-26 17:15:23.612 (UTC+0800)] info/eventloop     Stopping the EventLoop
[2024-06-26 17:15:23.612 (UTC+0800)] info/network       UDP     | Shutting down the ConnectionManager
[2024-06-26 17:15:23.612 (UTC+0800)] info/network       TCP     | Shutting down the ConnectionManager
[2024-06-26 17:15:23.613 (UTC+0800)] info/eventloop     The EventLoop has stopped

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4.0
 - [X] Other OPC UA SDKs used (client or server): Client
 - [X] Operating system: Windows 10
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-06-26:
In general it is a good idea to keep a session open.
The session might have existed prior and we try to activate an existing session for the new client.

But, if we just created the session ourselves and authentication fails, then we can also just delete it.
Any work towards this implementation is welcome.

yirongliang, 2024-06-27:
> 一般來說，保持會話開放是個好主意。 該會話可能之前已經存在，我們嘗試為新客戶端啟動現有會話。
> 
> 但是，如果我們自己建立了會話並且身份驗證失敗，那麼我們也可以將其刪除。 歡迎為此實施所做的任何工作。

[2024-06-26 17:29:28.697 (UTC+0800)] error/client Session cannot be activated with StatusCode BadIdentityTokenRejected. The client cannot recover from this, closing the connection.
[2024-06-26 17:29:28.700 (UTC+0800)] info/client Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected

Here the Session Status has been changed to Close. So I can think that the Client has issued a Session Close request to the Server, right?  However, the server does not seem to receive this request, so the session is not closed.

yirongliang, 2024-06-27:
I added a call to sendCloseSession in the responseActivateSession function.
This seems to solve my problem. I don't know if this is the right thing to do.

    UA_ActivateSessionResponse *ar = (UA_ActivateSessionResponse*)response;
    if(ar->responseHeader.serviceResult != UA_STATUSCODE_GOOD) {
        /* Activating the Session failed */
        /* Configuration option to not create a new Session */
        if(client->config.noNewSession) {
            UA_LOG_ERROR(client->config.logging, UA_LOGCATEGORY_CLIENT,
                         "Session cannot be activated with StatusCode %s. "
                         "The client is configured not to create a new Session.",
                         UA_StatusCode_name(ar->responseHeader.serviceResult));
            client->connectStatus = ar->responseHeader.serviceResult;
            cleanupSession(client);
            closeSecureChannel(client);
            UA_UNLOCK(&client->clientMutex);
            return;
        }

        /* The Session is no longer usable. Create a brand new one. */
        if(ar->responseHeader.serviceResult == UA_STATUSCODE_BADSESSIONIDINVALID ||
           ar->responseHeader.serviceResult == UA_STATUSCODE_BADSESSIONCLOSED) {
            UA_LOG_WARNING(client->config.logging, UA_LOGCATEGORY_CLIENT,
                           "Session to be activated no longer exists. Create a new Session.");
            cleanupSession(client);
            client->connectStatus = createSessionAsync(client);
            UA_UNLOCK(&client->clientMutex);
            return;
        }

        sendCloseSession2(client);
        cleanupSession(client);   

        /* Something else is wrong. Maybe the credentials no longer work. Give up. */
        UA_LOG_ERROR(client->config.logging, UA_LOGCATEGORY_CLIENT,
                     "Session cannot be activated with StatusCode %s. "
                     "The client cannot recover from this, closing the connection.",
                     UA_StatusCode_name(ar->responseHeader.serviceResult));
        client->connectStatus = ar->responseHeader.serviceResult;

        closeSecureChannel(client);
        UA_UNLOCK(&client->clientMutex);
        return;
    }

I modified sendCloseSession2 based on sendCloseSession. __Client_Service() cannot be called here because it will check the connection status and cannot send the request.