open62541/open62541 #6579: [Question] Event historial support

ivan2033, 2024-07-03:
We want to be able to read historical events from a PostgreSQL database through our OPCUA server. We are not sure if this is posible using the open62541 standard, as we have found a previous ticket in which it said it was not implemented yet.

[#2727](https://github.com/open62541/open62541/issues/2727)

Do you know if this is planned to be implemented soon?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-07-07:
The feature is implemented by a plugin .
The default implementation does not have it.
But the interface is there for you to add that.

ivan2033, 2024-07-09:
Ok, thanks for your answer. We are now in the right path. We have seen we can set a callback for readEvent and listen to the client history requests. 

Do you have an example/documentation of how to create the historyData that contains the events that we should return?