open62541/open62541 #6583: Cannot build open62541 with -DUA_ENABLE_ENCRYPTION=MBEDTLS on WSL

vovatrykoz, 2024-07-06:
## Description

I needed to build a version of open62541 with pubsub encryption on. However, when trying to build it I get the following CMake error:

```bash
[build] CMake Error at /usr/local/share/cmake-3.30/Modules/CMakeFindDependencyMacro.cmake:76 (find_package):
[build]   By not providing "FindMbedTLS.cmake" in CMAKE_MODULE_PATH this project has
[build]   asked CMake to find a package configuration file provided by "MbedTLS", but
[build]   CMake did not find one.
[build] 
[build]   Could not find a package configuration file provided by "MbedTLS" with any
[build]   of the following names:
[build] 
[build]     MbedTLSConfig.cmake
[build]     mbedtls-config.cmake
[build] 
[build]   Add the installation prefix of "MbedTLS" to CMAKE_PREFIX_PATH or set
[build]   "MbedTLS_DIR" to a directory containing one of the above files.  If
[build]   "MbedTLS" provides a separate development package or SDK, be sure it has
[build]   been installed.
[build] Call Stack (most recent call first):
[build]   /usr/local/lib/cmake/open62541/open62541Config.cmake:40 (find_dependency)
[build]   CMakeLists.txt:4 (find_package)
```
I understand that I might need to manually edit the FindMbedTLS.cmake file in order to let it find MbedTLS directories. However, it seems like CMake can't even find the FindMbedTLS.cmake.

I have installed libmedtls as per build instructions. Running
```bash
sudo apt-get install libmbedtls-dev
```

returns
```bash
libmbedtls-dev is already the newest version (2.28.0-1build1).
```
I had no problems building the code with ```-DUA_ENABLE_ENCRYPTION=OPENSSL``` however

I am trying to build the code in WSL, with Ubuntu 22.04.4 LTS

I used open62541 v. 1.4.2

## Background Information / Reproduction Steps

I run into this error when trying to build [pubsub_publish_encrypted.c](https://github.com/open62541/open62541/blob/d93776a049e116a5d1b24a7a3365abb2fe6fe94d/examples/pubsub/pubsub_publish_encrypted.c)

I have also tried building it in an essentially empty project

My own cmake looks like this

```
cmake_minimum_required(VERSION 3.20)
project(opcua_test)

find_package(open62541)

add_executable(opcua_run ${PROJECT_SOURCE_DIR}/source/main.cpp)
target_link_libraries(opcua_run open62541::open62541)

add_custom_target(run opcua_run)
```

main.cpp for the essentially empty project literally looks like this, there is no code there

```
#include <open62541/server.h>

int main(void) {
    return 0;
}

```

But as mentioned, I have also tried building [pubsub_publish_encrypted.c](https://github.com/open62541/open62541/blob/d93776a049e116a5d1b24a7a3365abb2fe6fe94d/examples/pubsub/pubsub_publish_encrypted.c) with the same CMake, using ```find_package(open62541)``` to locate open62541

Used CMake options:

```bash
cmake -DUA_ENABLE_PUBSUB_ENCRYPTION=OFF -DUA_ENABLE_ENCRYPTION=MBEDTLS ..
```
and
```bash
cmake -DUA_ENABLE_PUBSUB_ENCRYPTION=ON -DUA_ENABLE_ENCRYPTION=MBEDTLS ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-07-08:
We will look into that.
In the meantime, the documentation shows the preferred way to use open62541 with cmake and pkgconfig.

vkb, 2024-07-19:
The following works for me:

- Set up a custom cmake module path:
   e.g. set(CMAKE_MODULE_PATH ${CMAKE_SOURCE_DIR}/cmake/modules)
- Copy _[FindMbedTLS.cmake](https://github.com/open62541/open62541/blob/master/tools/cmake/FindMbedTLS.cmake)_ to set path
- Add _find_package(MbedTLS REQUIRED)_ to your CMakeLists.txt

jpfr, 2024-07-23:
This is now fixed on the 1.4 branch and part of the next patch release.

closed 2024-07-23 by jpfr