open62541/open62541 #6590: Client Subscriptions issue

polcontrol, 2024-07-09:
## Description

Hi,
I have a problem with the creation of subscriptions.

I can successfully connect to my test server from the UaExpert application, however when writing my own client I can only access the server data via pooling.

I am using the minimal examples from the open62541 repository.
My server is copied from `open62541/examples/server_mainloop.c`, and my client is from `open62541/examples/client_subscription_loop.c` (minor debug info added).

I see that the problem is with the following piece of client code:

```c
    case UA_SESSIONSTATE_ACTIVATED: {
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "A session with the server is activated");
        /* A new session was created. We need to create the subscription. */
        /* Create a subscription */
        UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "A request is created");
        UA_CreateSubscriptionResponse response =
            UA_Client_Subscriptions_create(client, request, NULL, NULL, deleteSubscriptionCallback);
        if(response.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                        "Create subscription succeeded, id %u",
                        response.subscriptionId);
        else {
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                        "Create subscription failed with error code: 0x%8.x",
                        response.responseHeader.serviceResult);

            return;
        }
```

The session with the server is activated, the request is created, the secure channel is open, but the subscribtion is failing:

```
[2024-07-09 11:46:22.951 (UTC+0200)] info/client        The initially defined EndpointURL opc.tcp://<server_name>:4840 is valid for the server
[2024-07-09 11:46:22.951 (UTC+0200)] info/client        Selected endpoint 0 in URL opc.tcp://<server_name>:4840 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-07-09 11:46:22.951 (UTC+0200)] info/client        Selected UserTokenPolicy open62541-anonymous-policy-none#None with UserTokenType Anonymous and SecurityPolicy
[2024-07-09 11:46:22.951 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2024-07-09 11:46:22.954 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2024-07-09 11:46:22.954 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2024-07-09 11:46:22.954 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2024-07-09 11:46:22.954 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2024-07-09 11:46:22.954 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2024-07-09 11:46:22.954 (UTC+0200)] info/userland      A session with the server is activated
[2024-07-09 11:46:22.954 (UTC+0200)] info/userland      A request is created
[2024-07-09 11:46:22.955 (UTC+0200)] error/eventloop    Cannot run EventLoop from the run method itself
[2024-07-09 11:46:22.955 (UTC+0200)] info/userland      Create subscription failed with error code: 0x80020000
[2024-07-09 11:46:23.962 (UTC+0200)] warn/client        Request with unknown RequestId 6
[2024-07-09 11:46:23.963 (UTC+0200)] error/client       Processing the message returned the error code BadSecurityChecksFailed
[2024-07-09 11:46:23.963 (UTC+0200)] info/channel       TCP 352 | SC 2  | SecureChannel closed
```

I have not found similar problems with the internal EventLoop on the internet.

I can see in the server logs that the subscription is being created, so the fault lies with the client.

```
[2024-07-09 11:46:22.954 (UTC+0200)] info/session       TCP 364 | SC 2  | Session "ns=1;g=efc92d72-4a4e-a3a6-d97d-97a9f61fe1be" | Session created
[2024-07-09 11:46:22.954 (UTC+0200)] info/session       TCP 364 | SC 2  | Session "ns=1;g=efc92d72-4a4e-a3a6-d97d-97a9f61fe1be" | ActivateSession: Session activated with ClientUserId ""
[2024-07-09 11:46:22.955 (UTC+0200)] info/session       TCP 364 | SC 2  | Session "ns=1;g=efc92d72-4a4e-a3a6-d97d-97a9f61fe1be" | Subscription 2 | Subscription created (Publishing interval 500.00ms, max 1000 notifications per publish)
[2024-07-09 11:46:23.964 (UTC+0200)] info/channel       TCP 0   | SC 0  | SecureChannel closed
[2024-07-09 11:46:23.964 (UTC+0200)] info/network       TCP 364 | Socket closed
```

How can I improve my code to make it work?

P. S. Thank you to everyone who contributed to this wonderful library!

## Background Information / Reproduction Steps

Used CMake options (powershell):

```powershell
cmake -DBUILD_SHARED_LIBS:BOOL="1" -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ARCHITECTURE:STRING="win32" -G "Visual Studio 17 2022" -S .. -B "build32"
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 11
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-07-12:
The problem is this: `Cannot run EventLoop from the run method itself`

Please search for this error message in the code, set a breakpoint and show us the stack trace.
It appears that you call a synchronous method (that needs to wait) from inside a callback that must not be waiting.

polcontrol, 2024-07-15:
This is my stack trace starting from the call of the callback function:

```
>	open62541.dll!UA_EventLoopPOSIX_run(UA_EventLoopPOSIX * el, unsigned int timeout) Line 304	C
 	open62541.dll!__Client_Service(UA_Client * client, const void * request, const UA_DataType * requestType, void * response, const UA_DataType * responseType) Line 627	C
 	open62541.dll!__UA_Client_Service(UA_Client * client, const void * request, const UA_DataType * requestType, void * response, const UA_DataType * responseType) Line 675	C
 	open62541.dll!UA_Client_Subscriptions_create(UA_Client * client, const UA_CreateSubscriptionRequest request, void * subscriptionContext, void(*)(UA_Client *, unsigned int, void *, UA_StatusChangeNotification *) statusChangeCallback, void(*)(UA_Client *, unsigned int, void *) deleteCallback) Line 117	C
 	test_open62541_client.exe!stateCallback(UA_Client * client, UA_SecureChannelState channelState, UA_SessionState sessionState, unsigned int recoveryStatus) Line 83	C++
 	open62541.dll!notifyClientState(UA_Client * client) Line 312	C
```

Indeed, as I create a subscription outside the callback function, everything works as expected.

It seems that in the example programme, calling UA_Client_Subscriptions_create() in the callback function causes EventLoop nesting which is detected in the `UA_EventLoopPOSIX_run()` function in `eventloop_posix.c`:

```c
static UA_StatusCode
UA_EventLoopPOSIX_run(UA_EventLoopPOSIX *el, UA_UInt32 timeout) {
    UA_LOCK(&el->elMutex);

    if(el->executing) {
        UA_LOG_ERROR(el->eventLoop.logger,
                     UA_LOGCATEGORY_EVENTLOOP,
                     "Cannot run EventLoop from the run method itself");
        UA_UNLOCK(&el->elMutex);
        return UA_STATUSCODE_BADINTERNALERROR;
    }
[...]
```

It appears that the example programme should be changed, as it now misleads users.

I am now wondering how reconnect to the server can be cleverly implemented in the current stituation (outside the callback function). Could anyone suggest a solution? As I find my ideas very awkward.

polcontrol, 2024-07-19:
I encountered another problem. The example states:

```c
    while(running) {
        /* if already connected, this will return GOOD and do nothing */
        /* if the connection is closed/errored, the connection will be reset and then reconnected */
        /* Alternatively you can also use UA_Client_getState to get the current state */
        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
```

This is not entirely true. If `UA_Client_connect()` is called when already connected, a warning is sent every cycle:

```
[2024-07-19 14:13:51.443 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.458 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.474 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.489 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.505 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.520 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.535 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.551 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.566 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.582 (UTC+0200)] warn/client        Client connection already initiated
[2024-07-19 14:13:51.597 (UTC+0200)] warn/client        Client connection already initiated
```

In the other hand, a call of `UA_Client_getState()` seems to return the connection status of the previous execution of the `UA_Client_connect()` function. Thus, even if the connection to the server is broken, it returns GOOD.

Am I missing something?