open62541/open62541 #6591: Exception when adding and receiving a Node, FreeRTOS

Maronari, 2024-07-11:
## Description
I need to run OPC server on Pi Pico using the FreeRTOS+lwIP architecture. I build sources like in #2511.  I get an exception when I try to add a VariableNode when creating a server in func UA_NODESTORE_NEW. Also, the server does not start because UA_Server_run is trying to get a Node in func UA_NODESTORE_GET, but an exception occurs. The program goes to crt0.S to the line "decl_isr_bkpt isr_hardfault".

OPC_task:

```bash
void opc_task(void *argument)
{
    UA_Boolean running = true;
    // Allows to set smaller buffer for the connections, which can cause problems
    UA_UInt32 sendBufferSize = 16000; // 64 KB was too much for my platform
    UA_UInt32 recvBufferSize = 16000; // 64 KB was too much for my platform
    UA_UInt16 portNumber = 4840;

    UA_ServerConfig *config;
    UA_StatusCode configStatus = UA_ServerConfig_setMinimalCustomBuffer(config, portNumber, 0, sendBufferSize, recvBufferSize);
    // VERY IMPORTANT: Set the hostname with your IP before allocating the server
    while (g_dns_get_ip_flag != 1)
    {
        vTaskDelay(50);
    }
    printf("[OPC UA] server IP: %s\n", ip4addr_ntoa(netif_ip4_addr(&g_netif)));
    UA_String UA_hostname = UA_STRING(ip4addr_ntoa(netif_ip4_addr(&g_netif)));

    //UA_String_clear(&config->customHostname);
    UA_String_copy(&UA_hostname, &config->customHostname);
    UA_Server *server = UA_Server_newWithConfig(config);

    // add a variable node to the adresspace
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 myInteger = 42;
    UA_Variant_setScalarCopy(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    attr.description = UA_LOCALIZEDTEXT_ALLOC("en-US", "the answer");
    attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en-US", "the answer");
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME_ALLOC(1, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_StatusCode status = UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                                parentReferenceNodeId, myIntegerName,
                                UA_NODEID_NULL, attr, NULL, NULL);

    if (status != UA_STATUSCODE_GOOD) {
        printf("UA_Server_addVariableNode() Status: 0x%x\n", status);
    }

    /* allocations on the heap need to be freed */
    UA_VariableAttributes_clear(&attr);
    UA_NodeId_clear(&myIntegerNodeId);
    UA_QualifiedName_clear(&myIntegerName);
    UA_StatusCode retval = UA_Server_run(server, &running);
    if (retval == UA_STATUSCODE_GOOD)
    {
        printf("[OPC UA] Server successfully running on %s:%d", config->customHostname.data, portNumber);
    }

    UA_Server_delete(server);
}
```
Stack trace:
```bash
isr_hardfault@0x100001c4 (RP2040_OPC_DHCP_C/libraries/pico-sdk/src/rp2_common/pico_standard_link/crt0.S:99)
<signal handler called>@0xfffffffd (Unknown source:0)
??@0x4f75678a (Unknown source:0)
AddNode_raw@0x10026e7c (RP2040_OPC_DHCP_C/port/open62541/open62541.c:42602)
Operation_addNode_begin@0x1002da22 (/RP2040_OPC_DHCP_C/port/open62541/open62541.c:42734)
Operation_addNode@0x1002daa4 (RP2040_OPC_DHCP_C/port/open62541/open62541.c:43120)
addNode@0x1002db48 (RP2040_OPC_DHCP_C/port/open62541/open62541.c:43179)
__UA_Server_addNode@0x1002dba4 (RP2040_OPC_DHCP_C/port/open62541/open62541.c:43199)
UA_Server_addVariableNode@0x1000055e (RP2040_OPC_DHCP_C/port/open62541/open62541.h:27625)
opc_task@0x1000055e (RP2040_OPC_DHCP_C/examples/opc_dhcp/w5x00_opc_dhcp.c:296)
xTimerCreateTimerTask@0x1000641c (RP2040_OPC_DHCP_C/libraries/FreeRTOS-Kernel/timers.c:282)
??@0x80000078 (Unknown source:0)
```

## Background Information / Reproduction Steps
Used CMake options:

```bash
cmake -DUA_ARCHITECTURE=freertosLWIP -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_GENERATE_NAMESPACE0=ON -DUA_ENABLE_DA=ON -DUA_NAMESPACE_ZERO=MINIMAL ../
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.11
 - [x] Other OPC UA SDKs used (client or server):  server
 - [x] Operating system: FreeRTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Maronari, 2024-07-17:
the error was in the incorrect initialization of the config

closed 2024-07-17 by Maronari

ritzi26, 2024-11-20:
I am now at the exact same point what was the problem with your config. And wich config do you mean? 
UA Server conf, FreeartosConfig.h or  lwipopts.h?

Maronari, 2024-11-22:
> I am now at the exact same point what was the problem with your config. And wich config do you mean? UA Server conf, FreeartosConfig.h or lwipopts.h?

Hello! As far as I remember, the issue was initializing the UA server config, this needs to be done after creating server, like
```
    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    retval = UA_ServerConfig_setMinimalCustomBuffer(config, portNumber, 0, sendBufferSize, recvBufferSize);
```

ritzi26, 2024-12-02:
Thanks for the answer.
Had it this way but my problem was actually in a stack overflow so somewhere else. Managed now to get it to run.