open62541/open62541 #6593: Logic how the ``UA_Server_run_iterate`` function works is not clear.

RavilN, 2024-07-13:
## Description

The function ``UA_Server_run_iterate`` is called with the argument ``waitInternal`` set to true. Internally it uses a hard-coded timeout of 200 ms to wait for events (network etc.). When no clients are connected, it takes roughly 200 ms to return, as expected in line with the hard-coded 200 ms timeout. It seems like by default when it is busy, the expected call frequency is every 1000 ms, because returned values are something like 800, 600, 400, that is, reminder from 1000 ms. But when the returned value becomes 0, the processing time becomes also 0 and it takes more than 20 calls until it again goes back to processing time taking roughly 200 ms, and the return values go to 800, 600, etc. So the sequence of returned values is 800, 600, 400, 200, 0, 0, 0, 0, 0, 0 ,0, 0, 0, 0, 0 (20or 30 times like this) ... 800, 600, 400,  etc.
The question is: what does it do in these more than 20 calls, when it seems like there is nothing to do? I would expect that processing time would stay around 200 ms (that is waits until timeout), and returned values would be like 800, 600, 400, 200, 0, 800, 600 and so on

## Background Information / Reproduction Steps

Measure the time between calls to the function ``UA_Server_run_iterate`` and print how long it took for the function to return and what is the return value.

Used CMake options: typical

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): ``v1.4.1``
 - [ ] Other OPC UA SDKs used (client or server):  n/a
 - [ ] Operating system: Ubuntu 22.04 running as WSL2
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached n/a
 - [ ] Wireshark network dump attached n/a
 - [ ] Self-contained code example attached n/a
 - [ ] Critical issue

jpfr, 2025-03-06:
This gets addressed in #7150.

jpfr, 2025-03-25:
Closed via #7102.