open62541/open62541 #6615: Generating the *.bsd file for custom datatypes

nictop17, 2024-07-26:
# Problem: Creating a  `*.bsd`  file for custom DataTypes
Hello,
I have a custom NodeSet that I created with the Nodeset-modeler SiOME and exported as `.xml`. The NodeSet contains custom DataTypes, such as the `BoxType` shown below:

To be able to use these datatypes with Open62541, the corresponding `*.bsd` file is required as far as I know (?). Unfortunately, I have not been able to find a sustainable (and open-source) way to create this (outdated) file. I have only been able to find an [old issue](https://github.com/open62541/open62541/issues/2784#issuecomment-499750909) from Profanter, but the approaches described are no longer feasible.
Is there a best practice for generating the `*.bsd`?
If so, why is the `*.bsd` file even still used in the implementation of Open62541 if it is deprecated and all necessary information are already available in the `.xml`?

Many Thanks for your help!

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v. 1.4.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

paperino1983, 2025-06-24:
normally you should be able to create the `bsd` file via the xml file generated by SiOME. When exporting the xml, click on export data type as node (obsolete)... It will generate the nodeset xml file with an added `TypeDictionary` node. So, follow [this](https://github.com/open62541/open62541/issues/2784#issuecomment-499750909) comment. In short copy that node's string, use some online base64 converter and then save the converted string as .bsd file.

o6-nbeier, 2026-10-20:
Thank you for opening this issue and for the details you provided.

It describes the same problem as the earlier issue #2587, so we are closing this one as a duplicate to keep the discussion in one place.

closed 2026-10-20 by o6-nbeier