open62541/open62541 #6650: Reconnect scenario doesn't work in Case of PubSub with MQTT connection

jchirantan, 2024-08-09:
## Description
- I am using Open62541 SDK for PubSub.
- I am trying to connect the subscriber to MQTT broker and fetch the Published values from broker.
- I am making multiple topic connections to MQTT broker from single PubSub connection.
- It works fine till the broker is up but as soon as broker goes down, the connection get terminated and PubSub goes in error state.
- It is not able to reconnect once the broker is up. 
- Is there any solution for how we can add multiple topic connections from single PubSub connection?

## Background Information / Reproduction Steps
- I am using the tutorial_pubsub_mqtt_subscribe.c file for testing this connection. https://github.com/open62541/open62541/blob/v1.4.2/examples/pubsub/tutorial_pubsub_mqtt_subscribe.c
and calling addReaderGroup method twice to create 2 readergroups (topic connections) in single PubSub connection.

2 connections established:
[image]

After the broker is stopped, the connection is closed and MQTT connections are also closed.
[image]

Used CMake options:

cmake.exe .. -G "Visual Studio 17 2022" -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Release -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_MQTT=ON -DUA_ENABLE_MQTT=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_DEBUG_SANITIZER=OFF -DUA_ENABLE_DISCOVERY_SEMAPHORE=OFF -DUA_ENABLE_HARDENING=OFF -DUA_ENABLE_NODEMANAGEMENT=OFF -DUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS=OFF -DUA_ENABLE_PARSING=ON -DUA_ENABLE_PUBSUB_DELTAFRAMES=OFF -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=OFF -DUA_ENABLE_STATUSCODE_DESCRIPTIONS=OFF -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=OFF -DUA_ENABLE_TYPEDESCRIPTION=ON -DUA_FORCE_WERROR=OFF -DUA_MSVC_FORCE_STATIC_CRT=OFF -DUA_ENABLE_ENCRYPTION_TPM2=OFF -DUA_LOGLEVEL=100 -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENABLE_PUBSUB_ENCRYPTION=ON

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): **v1.4.2**
 - [x] Other OPC UA SDKs used (client or server): **Prosys OPC UA Simulation Server**
 - [x] Operating system: Windows 
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2024-08-17:
Your connection goes into an error state.
We do not do automatic recovery from an error state.

You can just call UA_Server_enablePubSubConnection to attempt a reconnect.

Also you can register a callback in the config to be notified about any state changes for PubSub.

jchirantan, 2024-08-19:
@jpfr,

- Can we stop PubSub from going to error state somehow?

- Is there any way with which I can establish the connection containing multiple topics such that it will not go to error state when broker goes down for some time?

jchirantan, 2024-08-23:
Hi @jpfr,
Any update?

jchirantan, 2024-09-24:
Hi @jpfr 
Any progress on this?

jpfr, 2024-09-24:
This PR could haved fixed the problem:
https://github.com/open62541/open62541/commit/1c9116d6189953714f5ac14ddb851394ad0169e5

It is on 1.4 and master already.

jchirantan, 2024-10-21:
Hi @jpfr,
I tried implementing your changes also taken clone of latest release v1.4.6, but still issue is present.

- I am connecting multiple topics through single PubSub Connection.
- When I am closing the broker and restarting it again, the subscriber is not able to connect back.

anything else we can do here?

jchirantan, 2024-10-24:
Hi @jpfr 
any updates?

jchirantan, 2024-11-21:
Hi @jpfr, any updates?

jchirantan, 2024-12-06:
Hi @jpfr, any new updates on this issue?

jchirantan, 2025-01-07:
Hi @jpfr,
Any updates on this issue?

jchirantan, 2025-01-21:
Hi @jpfr,
Any updates?

jchirantan, 2025-02-10:
Hi @jpfr,
Any updates on reconnect scenario?

closed 2025-08-12 by jchirantan