open62541/open62541 #6654: [MQTT PubSub] Received Message does not update Subscribed DataSet Value

DocDriven, 2024-08-09:
## Description
I am trying to send messages via MQTT from one server (pub) to another server (sub). After struggling to see the received variables on the sub's side, I went to verify my approach by inspecting the examples. As far as I can tell, the examples have the same issue.

## Background Information / Reproduction Steps

I started the mosquitto service on default port 1883. I built and installed the latest master commit with examples and ran `tutorial_pubsub_mqtt_subscribe.c` and `tutorial_pubsub_mqtt_publish.c`. When inspecting the sent messages with the `mosquitto_sub` tool, I can see that the sent timestamp differs from message to message (bold):

> {"MessageId":null,"MessageType":"ua-data","PublisherId":2234,"Messages":[{"DataSetWriterId":62541,"SequenceNumber":23,"MetaDataVersion":{"MajorVersion":1069140163,"MinorVersion":1069139686},"Timestamp":"2024-08-09T14:50:06.0309774Z","Status":0,"Payload":{"Server localtime":{"Type":13,"Body":"**2024-08-09T14:50:06.0310016Z**"}}}]}
> 
> {"MessageId":null,"MessageType":"ua-data","PublisherId":2234,"Messages":[{"DataSetWriterId":62541,"SequenceNumber":24,"MetaDataVersion":{"MajorVersion":1069140163,"MinorVersion":1069139686},"Timestamp":"2024-08-09T14:50:06.5309733Z","Status":0,"Payload":{"Server localtime":{"Type":13,"Body":"**2024-08-09T14:50:06.5309967Z**"}}}]}
> 
> {"MessageId":null,"MessageType":"ua-data","PublisherId":2234,"Messages":[{"DataSetWriterId":62541,"SequenceNumber":25,"MetaDataVersion":{"MajorVersion":1069140163,"MinorVersion":1069139686},"Timestamp":"2024-08-09T14:50:07.0309739Z","Status":0,"Payload":{"Server localtime":{"Type":13,"Body":"**2024-08-09T14:50:07.0309958Z**"}}}]}
> 
> {"MessageId":null,"MessageType":"ua-data","PublisherId":2234,"Messages":[{"DataSetWriterId":62541,"SequenceNumber":26,"MetaDataVersion":{"MajorVersion":1069140163,"MinorVersion":1069139686},"Timestamp":"2024-08-09T14:50:07.5309709Z","Status":0,"Payload":{"Server localtime":{"Type":13,"Body":"**2024-08-09T14:50:07.5309925Z**"}}}]}

When inspecting the data on the subscriber side with the GUI tool `opcua-client`, the subscribed dataset was successfully created, however, the value of the marked variable is frozen at `1601-01-01T00:00:00`. To my understanding, this value should be updated when receiving a new message.

[image]

The messages are definitely received, because on every arrival the subscriber throws a warning:

> [2024-08-09 16:56:04.769 (UTC+0200)] warn/pubsub        PubSubConnection ns=1;i=52040   | ReaderGroup ns=1;i=50907      | Received a messaage for a non-operational ReaderGroup

I cannot tell if this is related to the issue, or if this is another issue. I figure that the relevant code is likely in the following section, which creates the nodes. But I cannot tell if and where the write from the received message to the local server field happens.
```
/**
 * **TargetVariables**
 *
 * The SubscribedDataSet option TargetVariables defines a list of Variable mappings between
 * received DataSet fields and target Variables in the Subscriber AddressSpace.
 * The values subscribed from the Publisher are updated in the value field of these variables */
    /* Create the TargetVariables with respect to DataSetMetaData fields */
    UA_FieldTargetVariable *targetVars = (UA_FieldTargetVariable *)
        UA_calloc(readerConfig.dataSetMetaData.fieldsSize, sizeof(UA_FieldTargetVariable));
    for(size_t i = 0; i < readerConfig.dataSetMetaData.fieldsSize; i++) {
        /* Variable to subscribe data */
        UA_VariableAttributes vAttr = UA_VariableAttributes_default;
        UA_LocalizedText_copy(&readerConfig.dataSetMetaData.fields[i].description,
                              &vAttr.description);
        vAttr.displayName.locale = UA_STRING("en-US");
        vAttr.displayName.text = readerConfig.dataSetMetaData.fields[i].name;
        vAttr.dataType = readerConfig.dataSetMetaData.fields[i].dataType;

        UA_NodeId newNode;
        retval |= UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, (UA_UInt32)i + 50000),
                                            folderId, UA_NS0ID(HASCOMPONENT),
                                            UA_QUALIFIEDNAME(1, (char *)readerConfig.dataSetMetaData.fields[i].name.data),
                                            UA_NS0ID(BASEDATAVARIABLETYPE),
                                            vAttr, NULL, &newNode);

        /* For creating Targetvariables */
        UA_FieldTargetDataType_init(&targetVars[i].targetVariable);
        targetVars[i].targetVariable.attributeId  = UA_ATTRIBUTEID_VALUE;
        targetVars[i].targetVariable.targetNodeId = newNode;
    }

    retval = UA_Server_DataSetReader_createTargetVariables(server, dataSetReaderId,
                                                           readerConfig.dataSetMetaData.fieldsSize, targetVars);
    for(size_t i = 0; i < readerConfig.dataSetMetaData.fieldsSize; i++)
        UA_FieldTargetDataType_clear(&targetVars[i].targetVariable);

    UA_free(targetVars);
    UA_free(readerConfig.dataSetMetaData.fields);
```
Can somebody verify that I am doing this correctly? What am I missing?

Used CMake options:

```bash
cmake .. \
          -DCMAKE_BUILD_TYPE=Release \
          -DUA_ENABLE_DA=ON \
          -DUA_ENABLE_DISCOVERY=ON \
          -DUA_ENABLE_ENCRYPTION=OPENSSL \
          -DUA_ENABLE_JSON_ENCODING=ON \
          -DUA_ENABLE_METHOD_CALLS=ON \
          -DUA_ENABLE_MQTT=ON \
          -DUA_ENABLE_PUBSUB=ON \
          -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON \
          -DUA_ENABLE_SUBSCRIPTIONS=ON \
          -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 22.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

DocDriven, 2024-08-12:
I enabled logging and messages hinting that the problem is related to empty attributes and the ReaderGroup:

> [2024-08-12 14:51:58.011 (UTC+0200)] info/eventloop     Starting the EventLoop
[2024-08-12 14:51:58.012 (UTC+0200)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-08-12 14:51:58.012 (UTC+0200)] info/server        AccessControl: Anonymous login is enabled
[2024-08-12 14:51:58.012 (UTC+0200)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-08-12 14:51:58.013 (UTC+0200)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=15303): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-08-12 14:51:58.015 (UTC+0200)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=25451): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50899
[2024-08-12 14:51:58.021 (UTC+0200)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50899): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50899
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50900
[2024-08-12 14:51:58.021 (UTC+0200)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50900): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50900
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50901
[2024-08-12 14:51:58.021 (UTC+0200)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50901): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50901
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50902
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50904
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50903
[2024-08-12 14:51:58.021 (UTC+0200)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50903): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50903
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50906
[2024-08-12 14:51:58.021 (UTC+0200)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50906): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50906
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50901
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50899
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50900
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50903
[2024-08-12 14:51:58.021 (UTC+0200)] info/pubsub        PubSubConnection ns=1;i=52040   | Connection created
[2024-08-12 14:51:58.021 (UTC+0200)] info/pubsub        PubSubConnection ns=1;i=52040   | State change: Disabled -> PreOperational
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50908
[2024-08-12 14:51:58.021 (UTC+0200)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;i=50908): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50908
[2024-08-12 14:51:58.021 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=1;i=50910
[... cut ...]
[2024-08-12 14:52:03.019 (UTC+0200)] warn/pubsub        PubSubConnection ns=1;i=52040   | ReaderGroup ns=1;i=50907      | Received a messaage for a non-operational ReaderGroup
[2024-08-12 14:52:03.519 (UTC+0200)] warn/pubsub        PubSubConnection ns=1;i=52040   | ReaderGroup ns=1;i=50907      | Received a messaage for a non-operational ReaderGroup
[2024-08-12 14:52:04.019 (UTC+0200)] warn/pubsub        PubSubConnection ns=1;i=52040   | ReaderGroup ns=1;i=50907      | Received a messaage for a non-operational ReaderGroup

Thiagorss54, 2025-01-08:
I am experiencing the same issue, and the proposed solution of adding UA_Server_enableDataSetReader did not work. I believe this function shouldn't be necessary since the UA_Server_enableAllPubSubComponents function is already being executed in the main.

francisjjp, 2025-01-31:
I have exactly the same issue. ReaderGroup is always in PREOPERATIONAL I have this:

`[2025-01-31 16:45:46.942 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2025-01-31 16:45:46.942 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2025-01-31 16:45:46.942 (UTC+0100)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2025-01-31 16:45:46.981 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | Connection created (State: Disabled)
[2025-01-31 16:45:46.982 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | ReaderGroup created (State: Disabled)
[2025-01-31 16:45:46.982 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | DataSetReader ns=1;i=54642 | DataSetReader created (State: Disabled)
[2025-01-31 16:45:46.982 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | DataSetReader ns=1;i=54642 | Disabled -> Paused
[2025-01-31 16:45:46.982 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Cannot enable the ReaderGroup while the server is not running -> Paused State
[2025-01-31 16:45:46.982 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Disabled -> Paused
[2025-01-31 16:45:46.982 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | Cannot enable the connection while the server is not running -> Paused State
[2025-01-31 16:45:46.982 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | Disabled -> Paused
[2025-01-31 16:45:46.982 (UTC+0100)] info/network       TCP     | Listening on all interfaces
[2025-01-31 16:45:46.983 (UTC+0100)] info/network       TCP 7   | Creating listen socket for "0.0.0.0" (with local hostname "orion78-lubuntu") on port 4840
[2025-01-31 16:45:46.983 (UTC+0100)] info/server        New DiscoveryUrl added: opc.tcp://orion78-lubuntu:4840
[2025-01-31 16:45:46.983 (UTC+0100)] info/network       TCP 8   | Creating listen socket for "::" (with local hostname "orion78-lubuntu") on port 4840
[2025-01-31 16:45:46.983 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | Paused -> Operational
[2025-01-31 16:45:46.983 (UTC+0100)] info/network       TCP 9   | Opening a connection to "192.168.0.23" on port 1883
[2025-01-31 16:45:46.983 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Attach receive-connection mqtt 9001
[2025-01-31 16:45:46.983 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Paused -> PreOperational
[2025-01-31 16:45:46.983 (UTC+0100)] info/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | DataSetReader ns=1;i=54642 | Paused -> PreOperational

**Conditions or Environment:**
[2025-01-31 16:45:46.983 (UTC+0100)] info/network       MQTT 9001       | Created connection subscribed on topic "datatopic"
[2025-01-31 16:45:55.286 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:45:56.283 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:45:57.300 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:45:58.294 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:45:59.299 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:46:00.283 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:46:01.283 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:46:02.297 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:46:03.282 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:46:04.347 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received a messaage for a non-operational ReaderGroup
[2025-01-31 16:46:05.284 (UTC+0100)] warn/pubsub        PubSubConnection ns=1;i=58192   | ReaderGroup ns=1;i=54636      | Received `

Compiled directly from master at date 01/31-2025 in lubuntu 64bits host.

**Cloning and preparing library:**

git clone https://github.com/open62541/open62541.git
cd open62541
git submodule update --init --recursive
mkdir build && cd build
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_SUSBCRIPTION=ON -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_NODEMANAGEMENT=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON - [... cut ...]
    UA_Server_enableAllPubSubComponents(server);
    UA_Server_runUntilInterrupt(server);

    UA_Server_delete(server);
    return 0;
}

Basically I have changed parameters:
1)  for connecting with my local mqtt mosquitto broker.
2) producer identified (coming from proxys, it is string)
3) id WriterGroup is 1.
4) id DataSetWriter is 1.
5) Modification for taking only a value in opua server. An int32 node.

I also put some screenshot with my simulated variables with proxys opc ua simulator pub sub.

[image]

[image]

[image]

francisjjp, 2025-02-03:
I am debugging in CodeBlocks and it appears MQTT message arrives correctly to OPC... 

[image]

francisjjp, 2025-02-03:
OK. Finally it seems I have found the issue or misunderstanding.

Program works OK when **static UA_Boolean useJson = false;**

The producer must sent data in **UADP** format through MQTT protocol.

[image]