open62541/open62541 #6657: Android Build with MBedTLS and OpenSSL Gives relocations in generic ELF error with crypto libraries

ffreality, 2024-08-14:
## Description
Open62541 gives these error while building for Android on Ubuntu with Android encryption libraries.

```
- (OpenSSL) libcrypto.a(a_int.o): Relocations in generic ELF (EM: 183)
- (MbedTLS) libmbedx509.a(x509.c.o): relocations in generic ELF (EM: 40)
```

## Background Information / Reproduction Steps

- I built mbedtls with `Unix Makefiles and Android NDK 25.2.9519653 android.toolchain.cmake`
- `Unreal Engine 5` has already `OpenSSL 1.1.1t` so there is additional build operation needs.
- I set CMake's generator as `Unix Makefiles with default native compilers`
- I set platform as posix
- CMake says configuration and generation done
- But terminal gives this error (I get same with openssl, too)
`/usr/bin/ld: /home/username/mbedtls/build/android/library/libmbedx509.a(x509.c.o): relocations in generic ELF (EM: 40)`

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 24.04 LTS (WSL2)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ffreality, 2024-08-15:
If I try to build `release v1.4.4` with `Android NDK 25.2.9519653 android.toolchain.cmake`, it gives `ld.lld: error: unable to find library -lrt` and -pthread error just like https://github.com/open62541/open62541/issues/5495

If I try to build `master` branch with `Android NDK 25.2.9519653 android.toolchain.cmake`, I need to comment out `#include <bits/stdio_lim.h>` from ua_certificategroup_filestore.c and `ua_securitypolicy_filestore.c`. After this, I could built library

**(this site stated that https://sites.uclouvain.be/SystInfo/usr/include/bits/stdio_lim.h.html using <stdio.h> will be better than bits/stdio_lim.h and actually both of these file stores have it.)**

BUT
Unreal Engine gives these errors.
```
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_types.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(types_generated.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(statuscodes.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_util.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_client.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_client_connect.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_client_discovery.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_client_highlevel.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_types_lex.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_types_encoding_json.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_config_default.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/OPC_Client/Source/OPC_Client/ThirdParty/Android/lib/libopen62541.a(ua_certificategroup_none.c.o) is incompatible with aarch64linux
```

jpfr, 2024-08-17:
For the stdio include, please open a PR with the change.

For the -lrt and -lpthread we might have to change the libraries we link against for Android.
Likewise you can modify the CMakeLists.txt and send a PR.

Make sure that you use the same compiler flags for open62541 and the rest of your application.
If you find an offending flag, we can just remove it from the CMakeLists.txt.

ffreality, 2024-08-18:
> For the stdio include, please open a PR with the change.
>
> For the -lrt and -lpthread we might have to change the libraries we link against for Android. Likewise you can modify the CMakeLists.txt and [...]

I did these to solve threading problems.
1. I disabled `-lpthread` and `-lrt` from `CMakeLists.txt`
2. I set `NDK_MIN_PLATFORM_LEVEL` from `ndk/build/cmake/platforms.cmake` as `24` because `open62541` uses `epoll_create1 (came with android-21) and <ifaddrs.h> for getifaddrs and freeifaddrs (came with android-24)`

Right now I have three scenarios.
1- If I use MBedTLS (custom build with same android_toolchain.cmake), `Unix Make on Ubuntu 24.04 LTS` can create `open62541.c`, `open62541.h` and `libopen62541.so` shared library without error. That shared file comes with ndk cmake toolchain but we don't have to use it. But Unreal Engine give ERROR 1 from below while packaging .apk file.

2- If I use OpenSSL from Unreal Engine, `Unix Make on Ubuntu 24.04 LTS` gives ERROR 2 from bellow.

3- If I disable encryption libraries, I can package Unreal Engine project and use OPC UA client but I can't make secure connections.

```
ERROR 1

UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(aes.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(ctr_drbg.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(entropy.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(error.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(md.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(pk.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(pkparse.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(pkwrite.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(rsa.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(sha1.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedcrypto.a(sha256.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedx509.a(x509.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedx509.a(x509_crl.c.o) is incompatible with aarch64linux
UATHelper: Packaging (Android (ASTC)): ld.lld: error: D:/Epic/Projects/FROZEN/SDKs/FF_Plugins_Mobile/Plugins/FF_MBedTLS/Source/FF_MBedTLS/ThirdParty/mbedtls/Android/lib/libmbedx509.a(x509_crt.c.o) is incompatible with aarch64linux
```

```
ERROR 2

ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(arm64cpuid.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(a_int.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(a_octet.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(a_time.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(d2i_pr.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(i2d_pr.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(bio_lib.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(bss_mem.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(bn_lib.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(err.o) is incompatible with armelf_linux_eabi
[... cut ...]
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(m_sigver.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(p_lib.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(pmeth_fn.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(pmeth_lib.o) is incompatible with armelf_linux_eabi
ld: error: /home/ffadmin/openssl/1.1.1t/lib/libcrypto.a(hmac.o) is incompatible with armelf_linux_eabi
```

ffreality, 2024-08-18:
Unreal can package open62541 plugin for android.

- I can connect a server with `NoSecure` mode
- see nodes with `UA_Client_forEachChildNodeCall`
- Read and Write nodes

But integrating encryption libraries cause these error and historical access crashes android application without error.

jpfr, 2024-08-19:
The encryption part looks to be specific for Unreal. Please ask in the unreal support channels. We cannot reproduce the problem here…

For the historizing crash, can you run this in a debugger? A backtrace of the crash helps fixing it.

ffreality, 2024-08-19:
> The encryption part looks to be specific for Unreal. Please ask in the unreal support channels. We cannot reproduce the problem here…
> 
> For the historizing crash, can you run this in a debugger? A backtrace of the crash helps fixing it.

I can understand you but OpenSSL problem doesn't come from Unreal Engine but Unix Makefiles. So, I ask this. What is correct workflow to build open62541 for arm64v8 android ?

ffreality, 2024-08-22:
> The encryption part looks to be specific for Unreal. Please ask in the unreal support channels. We cannot reproduce the problem here…
> 
> For the historizing crash, can you run this in a debugger? A backtrace of the crash helps fixing it.

I found the problem about historical access and It is not about function itself.

On Android, if I use same client for two operations at the same time (for example read and history), it causes crash.
I will put `Google AGDE (Android Game Development Extension)` result but I think it is about `-lpthread` and `-lrt.`

Also, if I use same client in both main thread and worker thread _(main thread is responsible for all mechanics of simulations. So, we use worker thread a.k.a FRunnableThread for reading hundreds or thousands of nodes at the same time)_ same crash happens again.

As I mention before, I commented out `-lpthread` and `-lrt` libraries and set android NDK level to 24 to automatically import them for NDK's itself. But it is not sufficient.

So, we have to find a better approach for building open62541 for Android.