open62541/open62541 #6658: Unable to generate amalgamated files for version 1.4.4

Pietro2018, 2024-08-17:
## Description
Moving from version 1.3.4 to 1.4.4 on LInux I am unable to generate amalgamated files on Raspberry PI.
Generation always terminates without open62541.c and open62541.h, while generates libopen62541.a.

- make 4.3
- cmake 3.25.1
- ccmake 3.25.1
- gcc 12.2.0

## Background Information / Reproduction Steps

I have tried to build using documented procedure: 

Building with CMake on Ubuntu or Debian[¶](https://www.open62541.org/doc/1.3/building.html#building-with-cmake-on-ubuntu-or-debian)

sudo apt-get install git build-essential gcc pkg-config cmake python

#enable additional features
sudo apt-get install cmake-curses-gui # for the ccmake graphical interface
sudo apt-get install libmbedtls-dev # for encryption support
sudo apt-get install check libsubunit-dev # for unit tests
sudo apt-get install python-sphinx graphviz # for documentation generation
sudo apt-get install python-sphinx-rtd-theme # documentation style

cd open62541
mkdir build
cd build

#selecting UA_ENABLE_ALAMGAMATION = ON
cmake .. 
make

I have also tried using ccmake with parameters, but without success.

cmake .. -DUA_ENABLE_AMALGAMATION=ON
make

Used CMake options:

```bash
cmake .. -DUA_ENABLE_AMALGAMATION=ON 
```

## Checklist

Please provide the following information:

 - [x ] open62541 Version 1.4.4 :
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ x] Operating system: Raspberry PI OS (bookworm) july 4th 2024, 32-bit 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2024-08-17:
Try `make open62541-amalgamation`.
This needs to be documented better.

Pietro2018, 2024-08-17:
Thank you. It works!

closed 2024-08-17 by Pietro2018