open62541/open62541 #6659: Multi-Threading is not yet supported for the client.

Pietro2018, 2024-08-17:
Multi-Threading is not yet supported for the client.
But you can use a custom mutex to ensure that only one thread can access the client at the same time.

_Originally posted by @jpfr in https://github.com/open62541/open62541/issues/3208#issuecomment-548440152_

Pietro2018, 2024-08-17:
Is this still true? I have searched for information about multithreading for client library, but I have not found any other information.

If the client library does not support multithreading, is it safe to solve the problem by maintaining a separate instance of `UA_Client` for each thread, without using a mutex?

jpfr, 2024-08-17:
Most of the client is thread-safe. If multithreading is activated in the build settings.

cf the `UA_THREADSAFE` annotations for client methods.

https://github.com/open62541/open62541/blob/126b67f5aa191fd0a3fa9ccb893374f0791bc331/include/open62541/client.h#L379

closed 2024-08-17 by jpfr