open62541/open62541 #6660: Can't change the opc server ip with linux bond ip when use the vision 1.3.5

YizhouEvan, 2024-08-19:
Due to work reasons, I had to use version 1.3.5 and could not upgrade to version 1.4. When I attempt to change the ip address in the following way (the ip address is 0.0.0.0), only the eth ip of Linux can be used for access, and the bond ip cannot be used for access.
- change the config.endpoint.endpointUrl
- change the config.netwprkLayers.discoveryUrl
- change the config.applicationDescription.applicationUri
- change the config.applicationDescription.discoveryUrls
- change the config.customHostname
Could anyone help me?

YizhouEvan, 2024-08-19:
In addition, when I directly use the bond ip address to replace the ip address, I cannot connect through the specified ip address and can only use the eth ip address.

jpfr, 2024-08-19:
What do you mean by eth ip?

YizhouEvan, 2024-08-19:
> What do you mean by eth ip?

the Ethernet
well Linux has a built-in driver for NIC bumding, which allows to bundle multiple physical NICs into separate
now I bundle eth0, eth1 into bond0.
When I create the opc server, I can only connect through the two ip addresses of eth and cannot connect through the ip addresses of bond. Even though I changed the customHostname to the bond ip, I couldn't access it.

closed 2026-10-20 by o6-nbeier