open62541/open62541 #6663: pthread and rt problem on Android

ffreality, 2024-08-23:
## Description

## Background Information / Reproduction Steps

As I mention on my encryption related problem, 
I made these changes to build open62541 for Android with Android NDK toolchain.
1. commented out `-lpthread` and `-lrt` libraries on `CMakeLists.txt`
2. commented out `<bits/stdio_lim.h>` in `ua_securitypolicy_filestore.c`
3. Add `set(ANDROID_PLATFORM 24)` to `CMakeLists.txt` 
Because `<ifaddrs.h>` (used by freeifaddrs and getifaddrs) and `epoll_create1` came with `NDK level 24`
4. set cross compile toolchain on cmake as `NDK 25.1.8937393 android.toolchain.cmake`
5. set platform type as Posix

I can build open62541, integrate it to Unreal Engine, package and run it.
But if I use same `UA_Client*` for two operations in different thread at the same time (for example read a node value and historical access), I get `art_sigsegv_fault` realted crash like this.

[image]

But Windows platform doesn't have this issue.

## Used CMake options:

```
USED NDK TOOLCHAIN = NDK 25.1.8937393 on Android Studio 2024.1.1 Patch 2 Koala on Ubuntu 24.04 LTS

-DMBEDTLS_INCLUDE_DIRS:PATH="/mbedtls/Win64/include" 
-DUA_ENABLE_ENCRYPTION_TPM2:STRING="OFF" 
-DCMAKE_INSTALL_LIBEXECDIR:PATH="libexec" 
-DUA_ENABLE_DISCOVERY:BOOL="1" 
-DUA_LOGLEVEL:STRING="100" 
-DUA_ARCHITECTURE:STRING="win32" 
-DUA_NAMESPACE_ZERO:STRING="FULL" 
-DCMAKE_INSTALL_SYSCONFDIR:PATH="etc" 
-DUA_ENABLE_PUBSUB_MONITORING:BOOL="1" 
-DUA_AMALGAMATION_MULTIARCH:BOOL="1" 
-DCMAKE_INSTALL_INFODIR:PATH="" 
-DUA_ENABLE_PUBSUB_ENCRYPTION:BOOL="1" 
-DUA_ENABLE_PUBSUB:BOOL="1"
-DUA_ENABLE_AMALGAMATION:BOOL="1" 
-DUA_ENABLE_DISCOVERY_MULTICAST:BOOL="1" 
-DUA_MULTITHREADING:STRING="100" 
-DCMAKE_INSTALL_SHAREDSTATEDIR:PATH="com" 
-DCMAKE_BUILD_TYPE:STRING="Release" 
-DCMAKE_INSTALL_BINDIR:PATH="bin" 
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4.4 release
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Development Platform = Ubuntu 24.04 LTS / Target Platform = Android
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ffreality, 2024-08-23:
I have a temporary solution like this.

```
UA_ReadResponse ReadResponse;
UA_ReadResponse_init(&ReadResponse);
ReadResponse = UA_Client_Service_read(In_Client, ValueReadRequest);
		
if (!ReadResponse.results)
{
	return false;
}
```

jpfr, 2024-08-24:
Please make this a separate issue each.
And wheee you have local fixes, PR are appreciated to speed things up.

closed 2024-08-28 by jpfr