open62541/open62541 #6669: Memory Leakage

SXiaoXiong, 2024-08-27:
## Description
Hi,
  We are developing a UA server with open62541. It worked perfectly until two weeks ago. 
  We found our program had memory leakage problem. At first, we checked our codes: to disable one component to check if the component leaks memory. Now, we focused on the open62541 library because we excluded other components by such exclusion method.
  
***
## Background Information / Reproduction Steps
  1. Create a 100k variables (int, float, string) server. And change all values of those variables per seconds.
  2. Use a UA client to connect to the server.
  3. Make the client subscribe 100k variables.
  4. Oberve the memory over 2 days.
  5. It shows about 10-20MB memory leakage.

***
## Used CMake options:
* UA_ENABLE_AMALGAMATION: TRUE
* UA_ENABLE_ENCRYPTION: OPENSSL
* UA_LOGLEVEL: 500
* UA_MULTITHEADING: 101
* BUILD_SHARED_LIBS: TRUE

***
## Checklist
 - [ ] open62541 Version (release number or git tag): 1.4.1
 - [ ] Operating system: Windows AMD64

jpfr, 2024-09-02:
Can you let your application run with valgrind or DrMemory?
That is (much) slower, but maybe a short runtime already exposes the origin of the leak.

SXiaoXiong, 2024-09-04:
We tried to detect the leakage by Deleaker. It shows the subscription function migth be the leakage source.
[image]