open62541/open62541 #6681: Client side certificates usage clarification

golovasteek, 2024-09-05:
## Description
According to example in https://github.com/open62541/open62541/blob/master/examples/client_connect.c
2 client-side certificates could be provided (-cert and -certAuth), and 2 corresponding private keys.
To my understanding, usually, the client has only 1 certifiacte+key pair.

Which one of these two should I use in typical case, and what is the use case for the other one? I didn't find any clarification in the docs or in the code...

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.4.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

neojaw, 2024-09-05:
One certificate+key pair (`-cert` in this case) is for creating an encrypted connection. The other one (`-certAuth`) is to log in and authenticate as a specific user, just like you would do with username/password.
Which ones you need depends on how the server manages connections and users, respectively.

uno1982, 2024-10-10:
I'm writing an unreal engine subsystem wrapper using this library and I'm also struggling on finding documentation on how to init the CertificationGroups or work within the new trustlist UA_TrustListDataType_init. Currently I'm reading the server .der and client .cer from the filesystem in my module directory and trying to create a group, 1 to store the server .der and add it to the trust list because I know this is valid and trusted via config. But I also need to pass the loaded .cer as part of the client config so its handed to the server when making the connection call  
// Connect to the server
 UA_StatusCode status = UA_Client_connect(Client, TCHAR_TO_UTF8(*ServerAddress));
 
If I can get beyond this point I'll simply trust the quarantined cert on the ignition server.

evgeny-fuse, 2024-10-11:
After trying to connect with different servers I found out that:
* you most probably would need the client certificate and the private key (it is a variable `certfile` in the example, parameter `-cert`, they are used for most of security policies. They don't have to be trusted (but I'm not sure if for all security policies non-trusted will work).
* the the `-certAuth` you may need only if the server administrator requires the auth with certificates

golovasteek, 2024-10-11:
After trying to connect with different servers I found out that:

    you most probably would need the client certificate and the private key (it is a variable certfile in the example, parameter -cert, they are used for most of security policies. They don't have to be trusted (but I'm not sure if for all security policies non-trusted will work).
    the the -certAuth you may need only if the server administrator requires the auth with certificates

neojaw, 2024-10-14:
@uno1982 There is two things that you can use certificates for: For creating a secure channel and for authenticating as a user. Both are (largely) independent from each other. You can do encryption with and without authentication and you can do authentication with and without encryption[^1].

* The server *security policies* determine whether secure and/or insecure connection is allowed. A security policy of `None` requires no encryption, a security policy of `Basic256Sha256` requires encryption with AES256 and SHA256. The server can have a list of security policies and the client can pick the one that suits best. Encryption needs certificates on both sides.
* The server *authentication mechanism* determines if you can "login" anonymously and/or and how you can authenticate as a user. For example in our product, you can login as anonymous with limited access (read only variables and only some methods callable). But you can also authenticate as one of a set of users with varying permissions. We do the authentication with username+password, but it could also be done with certificates.

[^1]: I think open62541 forbids password authentication over an unencrypted channel for security reasons.