open62541/open62541 #6684: Proposal to improve `UA_Server_addMethodNode` for FFI asynchronous support

NguyenVanDuongTDH, 2024-09-06:
Dear Open62541 developers,

I am currently using Open62541 with Dart FFI and have encountered an issue when trying to use asynchronous callbacks with the `UA_Server_addMethodNode` function. The current implementation only supports synchronous execution within the callback, which poses challenges when handling time-consuming tasks such as API calls or database access.

I would like to propose splitting this function into two parts:
1. A function to receive input from the client.
2. A separate function to handle the output, allowing asynchronous operations.

This change would provide more flexibility in implementing complex tasks and improve performance when asynchronous operations are needed.

Thank you, and I look forward to your feedback!

Best regards

jpfr, 2024-09-11:
Did you see that we support asynchronous methods?
This is a special flag that can be set for each method node.

NguyenVanDuongTDH, 2024-09-12:
> Did you see that we support asynchronous methods? This is a special flag that can be set for each method node.

i use dart ffi. ffi callback if i use dart async code it will cause error.
i saw UA_Server_setMethodNodeAsync function but i think it doesn't solve my problem

closed 2024-09-16 by NguyenVanDuongTDH