open62541/open62541 #6685: 1.4.4: Exception thrown when writing ByteString to Byte[]

tomitkl, 2024-09-06:
## Description

Exception is thrown in UA_Array_delete when the stack is trying to clear the write request after write.

That is probably because UA_ByteString is freed in adjustType function. At least, commenting out 

```
if(value->storageType != UA_VARIANT_DATA_NODELETE)
     UA_free(str);
```

seems to fix the issue.

The freeing part didn't exist in earlier version of adjustType. It has been added in https://github.com/open62541/open62541/commit/1ecbe1b3dba656696ac74ebfccafecab67be871f

## Background Information / Reproduction Steps

OPC UA Compliance Test Tool test case Attribute Services -> Attribute Write Values -> 007.js (Write: ByteString to Byte[]. Expects success.) can be run to reproduce the issue.

Used CMake options:

The stack is integrated to our own build system but default options should probably be OK to reproduce the issue.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4.4 (and probably master too)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tomitkl, 2024-09-06:
Call stack:

[image]

jpfr, 2024-09-11:
Thanks for the report.
We'll get this cleaned up for the upcoming patch release.

jpfr, 2024-09-11:
Fixed via #6695.

closed 2024-09-11 by jpfr